google-deepmind/mujoco #25: ./simulate is not working on RHEL 7.9

u6766669, 2021-10-21:
I am a novice user of Mujoco, I try to run the simulator on the remote cluster but I met the following problem. I tried some of the online solutions, but no help. Although it reports problems related to GLX, the command "glxinfo" and "glxgears" works well.

@mercury2 bin$ ./simulate 
mujoco pro version 2.10 x 
error of failed request: badrequest (invalid request code or no such operation) 
major opcode of failed request: 151 (glx) 
minor opcode of failed request: 0 () 
serial number of failed request: 9 
current serial number in output stream: 9 
segmentation fault

truatpasteurdotfr, 2021-11-14:
FYI: works for me on a remote CentOS-7 x86_64 using Nice DCV (https://download.nice-dcv.com/) with mujoco210-linux-x86_64.tar.gz.

saran-t, 2021-12-16:
Can you try again with the new 2.1.1 release?

kevinzakka, 2023-02-07:
Closing this for now. @truatpasteurdotfr feel free to open if you are still running into issues.

closed 2023-02-07 by kevinzakka