google-deepmind/mujoco #39: winch xml

flobotics, 2021-10-31:
Hi,
as the forums of mujoco are no longer available, we should ask here, so i do :) 

I want to describe a winch model with a steel-rope with mujoco xml. But i dont know what to use for a motor-actuator as joint ? In the simulator i can use the controls of "winch_0" but it does not rotate like a motor would  rotate the cylinder,  the rope moves up-and-down the cylinder ?

Here is my code so far:

```
<mujoco model="RopeWinch">

    
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
            width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <texture name="texcarpet" type="2d" file="carpet.png"/>
        <texture name="texsponge" type="2d" file="sponge.png"/>
        <texture name="texmarble" type="cube" file="marble.png"/>

        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
        <material name="matcarpet" texture="texcarpet"/>
        <material name="matsponge" texture="texsponge" specular="0.3"/>
        <material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
    </asset>

    <option timestep="0.002" jacobian="dense"/>

    <size nconmax="100" njmax="300" nstack="50000"/>

    <worldbody>
    	<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>
    	
        <body name="B0" pos="0 0 0.3">        
            <freejoint/>
            <body>
            	<geom type="cylinder" fromto="0 0 -0.3  0 0 0.3" size="0.06"/>
            	<joint name="winch_0" type="continous" pos="0 0 0.065" axis="0 1 0" range="-75 30" damping="5" stiffness="10" armature="0.02" />
            </body>
            <composite type="rope" count="40 1 1" spacing="0.04" offset="0 0 2">
                <joint kind="main" damping="0.005"/>
                <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
            </composite>
        </body>
    </worldbody>
    
    <actuator>
        <motor name="winch_0"       gear="200" joint="winch_0" />
    </actuator>    
</mujoco>`
```

It looks like this, where the rope goes up/down the cylinder. But the cylinder should rotate and the rope-connection-point should stay stationary.

[image]

What do i wrong ?

yuvaltassa, 2021-11-02:
Hi,

It is difficult to understand what you are trying to achieve, and your model had multiple issues with it. The model below does what I'm guessing you want. Compare to the one you sent to see the differences. I recommend familiarizing yourself with `simulate`'s visualisation options. One of the issues with your original model was that there were multiple self-collisions. The forces produced by these collisions can be visualised with the 'F' key. Transparency, enabled by the 'T' key is also useful here.

```
<mujoco model="RopeWinch">
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
            width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    </asset>
    
    <worldbody>
    	<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>
        <body pos="0 0 .3">
            <geom type="cylinder" fromto="0 0 -0.29  0 0 0.3" size="0.06"/>
            <joint name="winch_0" type="hinge" axis="0 0 1" range="-75 30" damping="5" stiffness="10" armature="0.02" />
            <body name="B0" pos=".06 0 0">                    
                <composite type="rope" count="40 1 1" spacing="0.04" offset="0 0 2">
                    <joint kind="main" damping="0.005"/>
                    <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
                </composite>
            </body>            
        </body>     
    </worldbody>
    
    <actuator>
        <motor name="winch_0" gear="200" joint="winch_0" />
    </actuator>    
</mujoco>
```

closed 2021-11-02 by yuvaltassa

flobotics, 2021-11-02:
hi,
thanks for your help and good tips. So far it works, but how do i connect more geoms to the rope?

I want a capsule that is connected in the middle of the rope to be fixed-connected, so that i can move it with the rope.

The winch works, but the red-capsule is not fix-connected to the rope, the rope falls away?

[image]

My code so far for the above:

`<mujoco model="RopeWinch">

    
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
        <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
            width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <texture name="texcarpet" type="2d" file="carpet.png"/>
        <texture name="texsponge" type="2d" file="sponge.png"/>
        <texture name="texmarble" type="cube" file="marble.png"/>

        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
        <material name="matcarpet" texture="texcarpet"/>
        <material name="matsponge" texture="texsponge" specular="0.3"/>
        <material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
    </asset>

    <option timestep="0.002" jacobian="dense"/>

    <size nconmax="100" njmax="300" nstack="50000"/>

    <worldbody>
    	<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>
    	
    	<body name="b_winch" pos="0 0 0.05">
	    	<geom type="cylinder" fromto="0 0 -0.05  0 0 0.05" size="0.06"/>
	    	<joint name="winch_0" range="-1.5708 1.5708"/>
		<geom type="cylinder" fromto="0 0 -0.04  0 0 -0.05" size="0.12"/>	
		<geom type="cylinder" fromto="0 0 0.04  0 0 0.05" size="0.12"/>
		
		<body name="B0" pos="0 0 0.00">
            		<composite type="rope" count="40 1 1" spacing="0.04" offset="0 0 2">
                		<joint kind="main" damping="0.005"/>
                		<geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
            		</composite>
        	</body>
        	
        	<body pos="0.5 0 0">
        	<geom type="capsule" size=".1 .15" rgba=".8 .2 .1 1"/>
	    	<joint name="winch_1" range="-1.5708 1.5708"/>
	    	</body>
	</body>     
    </worldbody>
    
    <actuator>
        <motor name="winch_0"       gear="2" joint="winch_0" />
    </actuator>    
</mujoco>`

Also i would like to connect the same winch-model to the other end of the rope, so that i can pull on both sides ?

rishabkr29, 2023-10-13:
Hey... Hii
I am also trying to do the similar thing, I wanted to wrap and unwrap the cable to acuate my manipulator. I was looking on your code but it is showing error.

yuvaltassa, 2023-10-13:
You need to save the model so that the composite (which is essentially a macro) gets expanded into the full tree. Then you can edit that model as you like.