google-deepmind/mujoco #40: Missing path to your environment variable (mujoco210)

bkpcoding, 2021-11-01:
Hey,
Today my license for mujoco expired so I downloaded the open-source version from DeepMind website "mujoco210" and placed it in ~/.mujoco folder and added the following line to my .bashrc file and source my .bashrc file. 
```
export LD_LIBRARY_PATH=/home/sagar/.mujoco/mujoco210/bin
export PATH="$LD_LIBRARY_PATH:$PATH"
```
But still, I get the following error:
```
Missing path to your environment variable. 
Current values LD_LIBRARY_PATH=/home/sagar/.mujoco/mujoco210/bin
Please add following line to .bashrc:
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sagar/.mujoco/mujoco200/bin
```
Can someone tell me if I am missing any point?
OS: Ubuntu 20.04 LTS
Python version: 3.7.5 
Conda version: 4.10.1

saran-t, 2021-11-01:
This is an issue with mujoco-py rather than MuJoCo itself. Our team has sent a [pull request to mujoco-py](https://github.com/openai/mujoco-py/pull/640) to upgrade it to work with 2.1 but so far their maintainers haven't responded. In the meantime, you should be able to proceed by patching the pull request locally.

closed 2021-11-01 by saran-t

bkpcoding, 2021-11-01:
This command did the job. First had to uninstall the previous mujoco-py version.
```
pip install free-mujoco-py
```