google-deepmind/mujoco #42: Build binary for linux aarch64

brandimarte, 2021-11-03:
Currently only Linux x86 binaries are available. It would be nice to have it for aarch64 architecture.

zyp57783, 2021-11-05:
The same question!

yuvaltassa, 2021-11-06:
We completely agree 👍 
Stay tuned.

Br3hman, 2021-11-10:
Looking forward to get 64 bit too. Mainly because I would like to try on windows subsystem Linux WSL. I was stupid enough to updated to windows 11 and I can't run any simulation anymore. I am not sure what is the connection but I will either switch back to windows 10 or use WSL.

saran-t, 2021-11-10:
@Br3hman just to be clear, this issue is about AArch64 (i.e. ARM) binaries. Our existing x86 binaries are already 64-bit.

Br3hman, 2021-11-10:
@saran-t thank you for pointing out. I will try to fix issue at my mind with WSL. It doesnt give me enough information to go on with it so far only say that "Illegal instruction". Maybe I am missing something. Sorry for spamming this thread.
Thanks 
Bilal

Chandanpanda, 2021-11-21:
Absolutely! Needed for Jetson platform...

closed 2021-12-16 by saran-t

saran-t, 2021-12-16:
AArch64 build is now available as part of our 2.1.1 release.

Br3hman, 2021-12-17:
This is great. Thanks, I will try it out today. Cheers.

Br3hman, 2021-12-18:
@saran-t in the next release could you also provide windows build binaries too as a wish list from my side.