google-deepmind/mujoco #43: howto add more geoms to a rope composite ?

flobotics, 2021-11-05:
hi,
i got a winch which which is connected to a rope, the only thing i found out till now is ,that i can control the position of the winch to the rope with the name attribute. yeah .sound strange, but if i use name="B0" , the winch is connected to the first capsule of the winch, if i use name="B10" its connected to the10th capsule (i think).   But how do i connect more than one geom or body to the rope ?

Here is a gif that  shows the winch that is connected to the first capsule of the rope (it seemed) and another red capsule that is NOT connected to the rope? How to connect the red-capsule to the rope ?

[image]

Here is my code:

```
<asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
        width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
    <texture name="texcarpet" type="2d" file="carpet.png"/>
    <texture name="texsponge" type="2d" file="sponge.png"/>
    <texture name="texmarble" type="cube" file="marble.png"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    <material name="matcarpet" texture="texcarpet"/>
    <material name="matsponge" texture="texsponge" specular="0.3"/>
    <material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
</asset>

<option timestep="0.002" jacobian="dense"/>

<size nconmax="100" njmax="300" nstack="50000"/>

<worldbody>
	<geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>
	
	<body name="b_winch" pos="0 0 0.05">
    	<geom type="cylinder" fromto="0 0 -0.05  0 0 0.05" size="0.06"/>
    	<joint name="winch_0" range="-1.5708 1.5708"/>
	<geom type="cylinder" fromto="0 0 -0.04  0 0 -0.05" size="0.12"/>	
	<geom type="cylinder" fromto="0 0 0.04  0 0 0.05" size="0.12"/>
	
	<body name="B0" pos="0 0 0.00">
        		<composite type="rope" count="40 1 1" spacing="0.04" offset="0 0 2">
            		<joint kind="main" damping="0.005"/>
            		<geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1"/>
        		</composite>
    	</body>
    	
    	<body pos="0.5 0 0">
    	<geom type="capsule" size=".1 .15" rgba=".8 .2 .1 1"/>
    	<joint name="winch_1" range="-1.5708 1.5708"/>
    	</body>
</body>     
</worldbody>

<actuator>
    <motor name="winch_0"       gear="2" joint="winch_0" />
</actuator>

 ```

thanks

yuvaltassa, 2021-11-06:
Please read the documentation regarding [composites](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-composite). The `<composite>` element is just a macro. If you **save the XML** from within `simulate`, the composite will be expanded and you can make any changes you like.

Note that there are two ways of connecting bodies. You can do in in the kinematic tree, or you can use [constraints](https://mujoco.readthedocs.io/en/latest/computation.html#coequality), these could behave differently, for example constraints can be made soft by setting the associated [solver parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#csolver).

closed 2021-11-06 by yuvaltassa

flobotics, 2021-11-06:
when i drag-and-drop my xml into simulate and then press the "save xml" button, nothing happens ?

vikashplus, 2022-04-08:
The saved xml can found in the same location as the simulate

yuvaltassa, 2022-04-08:
Hi @vikashplus :) 
Note this is different for MacOS, see #196
@kevinzakka has a WIP PR #213 to solve this which we'll likely merge post open-sourcing as it involves adding a dependency.