google-deepmind/mujoco #44: How to visualize contact force?

DonovanZhu, 2021-11-06:
Hi,
I am wandering how to visualize the contact force. I write a xml file and include the following line into it:

```
  <visual>
      <rgba com="0.502 1.0 0 0.5" contactforce="0.2 0.98 0.2 0.8" contactpoint="1.0 1.0 0.6 0.4"/>
      <scale com="0.2" forcewidth="0.05" contactwidth="0.10" contactheight="0.04"/>
  </visual>
```

Then I use the official file "simulate.cpp" to run this xml. I found that in the "randering" section, I can turn on the contact force and visualize the contact force as an arrow. Unfortunately, I still don't know how to show this contact force in my own program.  Does anyone have a clue?

Thanks!

erez-tom, 2021-11-06:
If you're using the built-in visualization machinery, you probably call
[mjv_updateScene](https://github.com/deepmind/mujoco/blob/main/include/mujoco.h#L594) in your rendering pipeline.

If, in this call, you pass [mjCAT_DECOR](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L29) (or mjCAT_ALL, as is done [here](https://github.com/deepmind/mujoco/blob/main/sample/simulate.cc#L1725)) as `catmask` (mjvGeom category mask) and make sure that the [flags](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L256) in your [visualization options](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L247) include [mjVIS_CONTACTFORCE](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L113), the visualizer code will read the [contact force](https://github.com/deepmind/mujoco/blob/main/include/mujoco.h#L450) and add the appropriate [visual geoms](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L197) to your [scene](https://github.com/deepmind/mujoco/blob/main/include/mjvisualize.h#L266) according to your visual specification as above.

closed 2021-11-06 by erez-tom

DonovanZhu, 2021-11-08:
Hi erez, thank you so much for your reply!
I tired to edit the mujoco init function by adding "opt.flags[mjVIS_CONTACTFORCE] = true;" after the "mjv_defaultOption(&opt);". The contact force dispaly successfully!
Thanks!

DonovanZhu, 2021-11-08:
Sorry for bothering you again, I found another problem in my program.
At frist, I run the "mj_step(m, d);" just like what it does in "basic.cpp". Like this:
```
    while( !glfwWindowShouldClose(window) )
    {
        // advance interactive simulation for 1/60 sec
        //  Assuming MuJoCo can simulate faster than real-time, which it usually can,
        //  this loop will finish on time for the next frame to be rendered at 60 fps.
        //  Otherwise add a cpu timer and exit this loop when it is time to render.
        mjtNum simstart = d->time;
        while( d->time - simstart < 1.0/60.0 )
            mj_step(m, d);

        // get framebuffer viewport
        mjrRect viewport = {0, 0, 0, 0};
        glfwGetFramebufferSize(window, &viewport.width, &viewport.height);

        // update scene and render
        mjv_updateScene(m, d, &opt, NULL, &cam, mjCAT_ALL, &scn);
        mjr_render(viewport, &scn, &con);

        // swap OpenGL buffers (blocking call due to v-sync)
        glfwSwapBuffers(window);

        // process pending GUI events, call GLFW callbacks
        glfwPollEvents();
    }
```
However, I want the "mj_step(m, d);" run in a stable 500Hz frequency loop so I can do some control algorithm for my simulated robot. I seperate the "mj_step(m, d);"  with other visulization parts into two differen thread. Then I found that the contact force arrows go creazy. They keep flashing and grow long in some frames.

Here is the visualization therad:

```
void mujoco_visualization_update(){

    // get framebuffer viewport
    mjrRect viewport = {0, 0, 0, 0};
    glfwGetFramebufferSize(window, &viewport.width, &viewport.height);

    // update scene and render
    mjv_updateScene(m, d, &opt, NULL, &cam, mjCAT_ALL, &scn);

    mjr_render(viewport, &scn, &con);

    // swap OpenGL buffers (blocking call due to v-sync)
    glfwSwapBuffers(window);

    // process pending GUI events, call GLFW callbacks
    glfwPollEvents();
}
```
Do you have any suggestion for this? Could mtx lock might help?
Thanks!