google-deepmind/mujoco #45: Unable to access mujoco forum

Ankur-Deka, 2021-11-06:
The link https://mujoco.org/forum/index.php doesn't seem to work anymore. Could you please make it accessible again? I found a discussion on a doubt at this link through google search only to realize that the link doesn't work anymore.

[image]

[image]

yuvaltassa, 2021-11-07:
In order to access the old forum, replace `mujoco.org` with `roboti.us` in your URL.
We might set up a redirect on `mujoco.org`, thanks for pointing this out.

closed 2021-11-07 by yuvaltassa

saran-t, 2021-11-07:
I've set up a redirect page on mujoco.org so that you're sent to the right place rather than 404s.

Ankur-Deka, 2021-11-10:
Thank you @yuvaltassa and @saran-t

pab47, 2022-08-01:
The older forum on this link does not seem to work anymore: https://roboti.us/forum/index.php
I tried different browsers (chrome/safari) with no luck. There was a wealth of information on that forum.

yuvaltassa, 2022-08-01:
Just delete the 's' from 'https', it works.

pab47, 2022-08-02:
Thanks. Just putting the complete linking for easy navigation. Please close the ticket.
http://www.roboti.us/forum/index.php