google-deepmind/mujoco #51: Feature request - Proximity queries

eaa3, 2021-11-09:
I was wondering if we currently have something akin to the distance and collision queries supported by Bullet [1] (e.g. getContactPoints, getClosestPoints). Is there any documentation on something equivalent to that in MuJoCo or could that be a potential feature to implemented in the future? Many thanks for the presentation at CoRL!

[1] - https://docs.google.com/document/u/0/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/mobilebasic#h.hrpbzrwzgy1p

yuvaltassa, 2021-11-10:
Hi, thanks for input! 

Regarding getContactPoints, simply iterate over the `data->contact` array to read out the contact points that have been found in the last call to the dynamics.

Regarding getClosestPoints, finding the distance and/or contact point between two arbitrary geoms -- independently of the dynamics -- there is currently no easy way to do this with the current API, but we can fairly easily add that. I've marked this as a feature request and will update here when we have something.

MotorCityCobra, 2021-11-29:
> simply iterate over the `data->contact` array to read out the contact points

Hello. In what function or part of the cc file would you do this?  
I've tried to print out a lot of the data with `d->` or `m->` and I get `Segmentation fault (core dumped)`

lzyang2000, 2023-10-27:
May I ask has there have been any updates on finding the distance between two arbitrary geoms? Thank you!

yuvaltassa, 2023-10-27:
Thanks for the reminder!

commit 02d01545 (2024-05-10): Add `mj_geomDistance` API function and sensors for geometric distance, normal direction and segment between between two geoms. Fixes #51

PiperOrigin-RevId: 632458301
Change-Id: I65b7e5bb0df59008028114cf4a07e0d6365ff3d3
closed 2024-05-10 by copybara-service[bot]