google-deepmind/mujoco #53: Current Contact Jacobian and dJacobian

dhruvthanki, 2021-11-09:
I want to access the value of the contact jacobian (J(q)) and its derivative (dJ(q,dq)) at the current state [q, dq] at a particular point on the robot. e.g the foot making contact with the ground. How do I go about doing this?

dhruvthanki, 2021-11-10:
I understand
```
int ncontacts = d->ncon;
const mjContact& c = d->contact[1];
            auto name1 = mj_id2name(m, mjOBJ_GEOM, c.geom1);
            auto name2 = mj_id2name(m, mjOBJ_GEOM, c.geom2);
```

yuvaltassa, 2021-11-10:
Hi, 

1. Regarding the contact Jacobian itself, this can be accessed as follows. Each contact in the `data->contact` array contains an [`efc_address`](https://github.com/deepmind/mujoco/blob/main/include/mjdata.h#L88) integer, which tells you what is the **first** row in the Jacobian that corresponds to this contact. The actual number of rows depends on the dimensionality of the contact (`condim`) and whether you are using elliptic or pyramidal contacts. If you are using sparse Jacobians you'll need to read the sparse format, if dense you can just read from [`efc_J`](https://github.com/deepmind/mujoco/blob/main/include/mjdata.h#L253). I recommend making a small test model which will default to dense (under 60 DoFs) and experimenting with elliptic and pyramidal. This is explained [here](https://mujoco.readthedocs.io/en/latest/computation.html#contact).
2. Regarding the derivative of the Jacobian, derivative with respect to what? It looks like, from your equations, that you want derivative with respect to time? For this you would need to finite difference, there is currently no way to compute dJ/dt analytically. I have not tried this but the following should work. At the end of `mj_forward()` (not `mj_step()`, since you don't want the state to change), save the Jacobian. Then call [`mj_integratePos`](https://github.com/deepmind/mujoco/blob/main/include/mujoco.h#L457), passing in `data->qpos` and `data->qvel`  with some tiny epsilon in the last argument to nudge the positions. Then call [`mj_makeConstraint`](https://github.com/deepmind/mujoco/blob/main/include/mujoco.h#L356) which computes the nudged Jacobian and fin-diff. Note that this will give you a possibly bad approximation of Jdot, since you are not recomputing the contact frames. You can do the nudging in the beginning of `forward()`, getting a better approximation (contacts will be recomputed), but then you run the risk of contacts appearing or disappearing, requiring some more logic.

closed 2021-11-10 by yuvaltassa

dhruvthanki, 2021-11-10:
Hi!
The corresponding contact force can be assessed using `mj_contactForce(m, d, i, result)` right?

yuvaltassa, 2021-11-14:
That is correct. Note that for elliptic cones this basically reads out the values from `efc_force` as the representation is the "natural" one. For pyramidal cones the normal and tangents need to be disambiguated.