google-deepmind/mujoco #54: Illegal instruction error when running under WSL (Windows)

saran-t, 2021-11-10:
@saran-t thank you for pointing out. I will try to fix issue at my mind with WSL. It doesnt give me enough information to go on with it so far only say that "Illegal instruction". Maybe I am missing something. Sorry for spamming this thread.
Thanks 
Bilal

_Originally posted by @Br3hman in https://github.com/deepmind/mujoco/issues/42#issuecomment-965095586_

saran-t, 2021-11-10:
@Br3hman what are the specs of the machine that you're trying to run this on? Specifically, what's the CPU model?

Br3hman, 2021-11-10:
OS Name	Microsoft Windows 11 Home
Version	10.0.22000 Build 22000
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	SURFACEGO
System Manufacturer	Microsoft Corporation
System Model	Surface Go 2
System Type	x64-based PC
System SKU	Surface_Go_2_1901
Processor	Intel(R) Pentium(R) CPU 4425Y @ 1.70GHz, 1704 Mhz, 2 Core(s), 4 Logical Processor(s)
BIOS Version/Date	Microsoft Corporation 1.0.16, 24/05/2021
SMBIOS Version	3.0
Embedded Controller Version	1.03
BIOS Mode	UEFI
BaseBoard Manufacturer	Microsoft Corporation
BaseBoard Product	Surface Go 2
Boot Device	\Device\HarddiskVolume1
Hardware Abstraction Layer	Version = "10.0.22000.1"
Username	SURFACEGO\user
Time Zone	W. Europe Standard Time
Installed Physical Memory (RAM)	8.00 GB
Total Physical Memory	7.87 GB
Available Physical Memory	745 MB
Total Virtual Memory	14.9 GB
Available Virtual Memory	5.09 GB
Page File Space	7.00 GB

saran-t, 2021-11-10:
Your CPU [doesn't support the AVX instruction set](https://ark.intel.com/content/www/us/en/ark/products/192786/intel-pentium-gold-processor-4425y-2m-cache-1-70-ghz.html). Were you somehow able to get it to work on the same machine under Windows 10? This shouldn't have been possible.

We can consider building without AVX as well to support this use case.

Br3hman, 2021-11-10:
Sorry for mixed up, it was another machine I ran it. And I thought maybe it's not working on surfaces go because of the windows update. Now I know thanks for clearing up. Best regards
Bilal

Br3hman, 2021-11-10:
Of you could build without AVX that would be great. Let me know if I can try it for you.

zyp57783, 2021-11-15:
I am looking for non-AVX build too, for some specific processors like apple M1 which does not support AVX.

saran-t, 2021-11-15:
For Apple Silicon we'll be providing a AArch64 build (or rather a Universal Binary) so AVX isn't a concern there. See Issue #2

saran-t, 2022-05-23:
We are not going to provide prebuilt binaries without AVX, however since MuJoCo is now open source people should be able to build these for themselves if required.

closed 2022-05-23 by saran-t