google-deepmind/mujoco #55: Connecting two body objects to the child body in the kinematics tree

rokas-bendikas, 2021-11-12:
Hello, I am making a custom model, which I am converting from URDF. In the original urdf model, there are two objects which are connecting to the same child body, however mujoco transated that into one of the bodies being connected and the other one is open-ended. Is there some sort of trick to have two bodies having the same child in the kinematics tree? I am refering to psm_pitch_top_link and psm_pitch_bottom_link both being connected to psm_pitch_end_link in the attached model. Thanks!

[psm.txt](https://github.com/deepmind/mujoco/files/7527801/psm.txt)

saran-t, 2021-11-12:
The problem here is that you have a kinematic loop rather than tree. You'll want to define an [equality constraint](https://mujoco.readthedocs.io/en/latest/computation.html#equality) between `psm_pitch_top_link` and `psm_pitch_end_link`.

rokas-bendikas, 2021-11-18:
It seems like the only way to connect two bodies outside the kinematic tree without welding them is the 'connect' equality constraint. I used solref="0.002 1" to stiffen the joint.

closed 2021-11-18 by rokas-bendikas