google-deepmind/mujoco #57: Strange behavior in wheel slip simulation.

silverjoda, 2021-11-14:
Hello, 
I'm currently working on a sim-to-real project which requires a simulation of a rear-wheel driven car
on a low-friction surface (meaning the car drifts and slips a lot). I'm getting strange behavior where when turning, the car crabs 
to one side, as if there was a parasitic force acting on it, as on the video.  Is this a bug or is it a matter of setting simulator parameters correctly? I tried various things and nothing works. The simulation is from the [Mushr](https://mushr.io/tutorials/mujoco/) tutorial. I only decreased the friction of the wheels of the car and I can't get it to work as one would expect it to.

https://user-images.githubusercontent.com/21022992/141686716-156325f6-724e-4f94-9962-778fcc2a8f70.mp4

yuvaltassa, 2021-11-14:
Hi,
Could you please provide a link to your model?

silverjoda, 2021-11-14:
> Hi, Could you please provide a link to your model?
[mj_models.zip](https://github.com/deepmind/mujoco/files/7534373/mj_models.zip)

The one_car.xml includes the buddy.xml  
I tried playing with all sorts of parameters such as solimp, solref, solver type, integrator, timestep, condim
and others but I can't get rid of this behavior. It works fine if you set it to a high enough friction so that it doesn't slip then the wierd stuff starts happening.

Thanks in advance.

yuvaltassa, 2021-11-14:
Hi,
Could you please add the mesh directories and otherwise make sure the model loads?
Thanks

silverjoda, 2021-11-14:
[models.zip](https://github.com/deepmind/mujoco/files/7534506/models.zip)
I am sorry, I did not think that you would run it. Here is the entire models directory. 
It's taken from this repo: https://github.com/prl-mushr/mushr_mujoco_ros

yuvaltassa, 2021-11-14:
Hi,
Thank you for the interesting problem. This model is badly designed, but in a very subtle way.

- First, how to diagnose:
Load the model (I loaded "one_car.xml") in `simulate` and hit the **C** button to activate contact visualisation. Now zoom in (mouse wheel), then hold the **Control** key and right-button double-click on the car to make the floating camera track the car. Then use the sliders on the right to navigate your car. You should be able to see contacts turning on and off. Why are they doing this? 
Look closely at one of the wheels. Each wheel has two geoms (defined in "buddy.xml"), one is a translucent, non-colliding cylindrical mesh, the other is a primitive ellipsoid which collides with the floor. But how is the ellipsoid defined? It is defined by the line `<geom fitscale="1.2" type="ellipsoid" friction="0.4 0.005 0.0001" condim="6" contype="1" conaffinity="0" mesh="buddy_mushr_wheel" mass="0.498952" />`, which is telling the MuJoCo compiler to **fit an ellipsoid** to the mesh called "buddy_mushr_wheel". Because there is something about this specific mesh which is _radially non-isotropic_, the fitted ellipsoid is _not precisely aligned with the axle_, only almost-aligned. As the wheel turns it loses and regains contact with the ground leading to the effect you observed.

- Second: how to solve.
There are multiple ways of solving this. The most realistic solution is to give the wheels some sort of suspension system, after all, maintaining contact with the ground is precisely what suspension is for. In fact, the mesh of the original chassis has suspension, but it is not actually implemented in your model. A quicker solution is to replace the fitted ellipsoids with properly centred ellipsoids. E.g. the following geometry (wheels with 5cm radius) seems to do the trick:
```
<default class="buddy_wheel_collision">
    <geom type="ellipsoid"  size=".05 .02 .05"  .../>
</default> 
```
You will still see the contacts flashing out intermittently, though not like before, but that is because any non zero roll or pitch of the four co-planar contact points will lead to one of them not touching the ground. The correct way to solve this (and indeed this will also work for the original model, but unbalanced wheels are probably a bad idea in any case), is to _soften the contacts_. This is how one models soft things (like a rubber tyre) in MuJoCo. You can do this directly in `simulate` by enabling "override" (under "physics"), which will let you override all contact parameters in the model. In the model with the perfectly aligned wheels (using the snippet above), it was enough to set `solimp[0]=0`. To understand this better, please read up on [contact parameters](https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters).

Please let us know if you have any further questions. 

One last thing, I would appreciate if you forwarded this answer to the Mushr author, perhaps they will fix their model.

Cheers.

closed 2021-11-14 by yuvaltassa

silverjoda, 2021-11-15:
Thanks a lot for your answer. I think that I understand the solution.
I tried replacing the buddy_wheel with the buddy_wheel_collision class that you
have suggested and changed solimp[0] but I'm still not getting the drifting behavior that I would like when turning.
I think that the core issue is the lack of suspension as you mentioned because I can see the contacts bouncing from 
left to right, albeit not as erratically as before. I guess I'll try to add a spring/damper element to each wheel and see what happens.

In any case, thanks a lot, it was definitely useful.
T.

silverjoda, 2021-11-24:
How to solve: 
If anyone is interested in the solution, I used slider joints to add simple spring/damper suspension for each wheel, used an ellipsoid collision for each wheel and added solimp[0] = 0 on wheel collisions and the floor geometry and it works as expected now. Here is the updated models directory if anyone is interested
[models.zip](https://github.com/deepmind/mujoco/files/7594489/models.zip)
.