google-deepmind/mujoco #58: Mujoco MATLAB integration

AlKramb, 2021-11-15:
Hi, 
For some years I was using the Mujoco Haptix version, which comes with a precompiled MATLAB interface library. The version is not available on your web page or git, therefore I am curious if there are any plans for the future, for the MATLAB interface library to be available as open-source for Mujoco?
Thank you for the reply in advance!

yuvaltassa, 2021-11-15:
Hi,
We do not use Matlab ourselves, but we could certainly consider open-sourcing the plug-in.
Would you be open to maintaining it as an open-source contributor, or do you just want to use it?

AlKramb, 2021-11-16:
Hi gain, thank you for your reply, just to make it clear, is there a Matlab plug-in that can be used with Mujoco (not Haptix)? I personally would be interested in using it, but I will check with my colleague if he would be interested in maintaining it because he is very involved with the Matlab development. 
Br.

yuvaltassa, 2021-11-16:
A Matlab mex plug-in is available for _MuJoCo 2.0 for Windows_ (`mjx.cpp` in the `sample/` directory) and can be found in the roboti.us [legacy downloads page](https://www.roboti.us/download.html). We currently do not support this plug-in, but we could consider helping external maintainers who wish to support it.

yuvaltassa, 2022-05-23:
Hi,

Let us know if you've made any progress with this, we'd be happy to help.
Closing this issue in the meantime.

closed 2022-05-23 by yuvaltassa