google-deepmind/mujoco #59: Build binary for linux ppc64le

rodrigodelazcano, 2021-11-15:
Is it possible to port the x86 binaries to Linux IBM POWER architecture?

saran-t, 2021-11-15:
In principle we can cross-compile MuJoCo since the code is portable C, but we won't be able to test (not even basic smoke test) that it works since we don't have the hardware.

Can you tell us a bit more about your intended use case?

traversaro, 2021-11-16:
Just FYI for anyone interested in ppc64le binaries, at https://conda-forge.org/ we automatically provide ppc64le packages for many robotics simulation related packages, see for example (if you check under platforms, you will see `linux-ppc64le`):
* https://github.com/conda-forge/dartsim-feedstock
* https://github.com/conda-forge/bullet-feedstock
* https://github.com/conda-forge/libignition-physics-feedstock/

This packages are obtained either via compilation on Travis (https://blog.travis-ci.com/2019-11-12-multi-cpu-architecture-ibm-power-ibm-z) or cross-compilation and tests run on QEMU in Azure. 

Note that this packaging effort is complementary to any official binary provided upstream, but I tought it was relevant to mention this here as the plan is to package mujoco in conda-forge as soon as the source code is available, and in that case (unless the build system is strange in someway, i.e. not a common autotools/meson/cmake one) we could get ppc64le binaries with little effort.

/fyi @Tobias-Fischer @wolfv

rodrigodelazcano, 2021-11-16:
> In principle we can cross-compile MuJoCo since the code is portable C, but we won't be able to properly test that it works since we don't have the hardware. (QEMU might be a viable option if there's a good reason for us to go down that route)
> 
> Can you tell us a bit more about your intended use case?

Hi Saran, thank you for your prompt response. I'm trying to train an RL application in a distributed system. For that I want to deploy at once the same Mujoco simulation  in multiple nodes of an HPC . The problem is that the HPC nodes have IBM POWER9 cpu's with ppc64le architecture. I'm not very familiar with cross-compilation, but having an initial make file that works on QEMU or Travis as suggested by @traversaro would be of great help.

Correct me if I'm wrong, but currently I can't compile-test Mujoco myself since it is a pre-built library right?

saran-t, 2021-11-16:
@rodrigodelazcano correct, until we open source MuJoCo any additional platform will require us to set up new build on our side. I'll look into ppc64le and if it's not a great deal of extra work we can make it part of our CI.

After open sourcing I'd be happy to have the ppc64le build be owned and maintained by someone else though :)

saran-t, 2021-11-25:
@rodrigodelazcano Do you need to be able to render on the PPC machine? Is it sufficient if we provide the "no_gl" binary for this platform?

rodrigodelazcano, 2021-11-25:
> @rodrigodelazcano Do you need to be able to render on the PPC machine? Is it sufficient if we provide the "no_gl" binary for this platform?

@saran-t yes, no need to have rendering feature. Thanks!

closed 2021-12-16 by saran-t

saran-t, 2021-12-16:
ppc64le build is now available as part of our 2.1.1 release, with and without rendering support. Basic tests passed on qemu (ppc64-softmmu).

We can't offer any support for this, though.