google-deepmind/mujoco #60: Could anyone please explain scale and units of measure in MuJoCo?

MotorCityCobra, 2021-11-17:
I've seen Emo Todorov comment before that MuJoCo assumes no units of measurement and objects are just the size they are, or something to that effect. But I'm still left with questions.  
When bodies are falling to the ground should the scale factor in some way?  

If the meshes of humanoid parts combine to be the size of a human in Solidworks does that mean the model is 165 cm high in MuJoCo and falls with the force and speed just like a 165 cm high humanoid bot falls in real life?  
I could also use a better guide of size and position. Thank you!

closed 2021-11-17 by MotorCityCobra

yuvaltassa, 2021-11-17:
The units of length are indeed undefined, you are free to interpret them as you like (metre, centimetre, inch etc.)

People very often add a layer of colliding primitive geoms alongside their non-colliding meshes (which are used for visualisation and inertial properties), since primitive collisions are faster to compute than mesh collisions.

MotorCityCobra, 2021-11-17:
> The units of length are indeed undefined

For the position, `pos` in '<geom name="link1" type="mesh" mesh="leg" pos="0 0 180"/>`   
What is the `180`? What is it 180 of?