google-deepmind/mujoco #61: I'm learning STL file / objects in XML models are convex hulls. Does this mean it's not possible to model real robots?

MotorCityCobra, 2021-11-17:
Because if you can't get the proper volume and density into your physics simulator how can you simulate it?  

I was hoping to source the CAD files of parts I was going to use, assemble them all in Solidworks and simulate the model in MuJoCo? Would I have to give the make the mesh (STL) imports zero mass to bring them into MuJoCo? Real CAD Joints and motors won't fit together if they are convex hulls. RIght?

yuvaltassa, 2021-11-17:
MuJoCo will estimate masses and inertias of your STL given a density, or you can use the possibly more precise values given by your CAD software using the [inertial](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-inertial) clause. 

Convex hulls are used for collision detection. Note that (by default) parent and child bodies don't collide, so there is usually no problem of "not fitting together".

closed 2021-11-17 by yuvaltassa

MotorCityCobra, 2021-11-17:
> there is usually no problem of "not fitting together".

I'm a little new to this, so maybe I misunderstand. If you have STL meshes of a nut and bolt imported to MuJoCo will the bolt be able to be in a position where it has been screwed into the nut? As I understand it a convex hull wraps the nut and makes it impenetrable. As I understood it the way of dealing with this was to give the mesh objects no mass and put other shapes in and around, the nut and bolt in this case.  
Thank you for the response!

Hobbit-wgh, 2023-07-21:
> 
I meet the same problem. I want to simulate the  bolt screwed into the nut, but I can not get correct result. 

I would like to consult with the senior.

yuvaltassa, 2023-07-23:
Search the issues for CoACD

sunnymints, 2023-08-19:
> > 
> 
> I meet the same problem. I want to simulate the bolt screwed into the nut, but I can not get correct result.
> 
> I would like to consult with the senior.

Hello, my senior, I also want to simulate the bolt preload in mujoco , but I'm not sure how to correct it.

yuvaltassa, 2023-08-21:
[MuJoCo latest](https://mujoco.readthedocs.io/en/latest/changelog.html) now supports Signed Distance Functions for collisions. See e.g. [this video](https://youtu.be/QewlEqIZi1o)