google-deepmind/mujoco #64: Exceeded joint limit range

rokas-bendikas, 2021-11-17:
I observed that some of my hinge joints exceed the joint limits that are specified using 'range' parameter whilst setting 'limited' flag as 'true'. What could be the possible reasons for this?

yuvaltassa, 2021-11-17:
The reason is that in MuJoCo all constraints are soft. This is by design,
and related to dynamics invertibility. You can stiffen (or soften) the
limit constraints using the joint's solimplimit and solreflimit parameters,
but you cannot make them infinitely stiff (such that there will never be
any violation).

Note that this design choice is also realistic, as infinitely stiff joint
limits are physically impossible.

closed 2021-11-23 by rokas-bendikas

Krishnendu1984, 2022-08-21:
I have made a sliding joint but with joint range specified and limited set as true but still while using actuator in GUI it is exceeding the joint range limit.How to adjust that?

yuvaltassa, 2022-08-21:
Read the reply above.

Krishnendu1984, 2022-08-21:
yes sir I have set the solimplimit =1000 1000 1000 but at the extreme joint ends the body is oscillating there ,it should stop at that poistion

Krishnendu1984, 2022-08-21:
position without oscillation

yuvaltassa, 2022-08-21:
That is not how `solimplimit` works, please read the related documentation.

https://mujoco.readthedocs.io/en/latest/modeling.html#csolver