google-deepmind/mujoco #65: Is the python library mujoco-py==2.1 needed to run this latest version of mujoco 2.1?

matte-esse, 2021-11-18:
I (pip) installed the latest version of `mujoco-py==2.0.2.13` and it keeps looking for mujoco200 path. Is this expected behavior? Is this new 2.1 version still not actually usable?

floringogianu, 2021-11-19:
Don't know if this answers your question, I was able to install `mujoco-py 2.1` just now with:

```
sudo apt install libosmesa6-dev libgl1-mesa-glx libglfw3 libgl1-mesa-dev libglew-dev
pip install -U 'mujoco-py<2.2,>=2.1' --no-cache-dir --no-binary :all: --no-build-isolation
```
The previous `mujoco-py` versions are indeed looking for the wrong mujoco versions.

matte-esse, 2021-11-19:
with the release of https://pypi.org/project/mujoco-py/2.1.2.14/ this issue is solved

closed 2021-11-19 by matte-esse