google-deepmind/mujoco #71: Are there any time estimates for when the entire source code will be released?

MotorCityCobra, 2021-11-22:
From the Read Me...
>we are currently making preparations to open source the codebase. In the meantime, MuJoCo is available for download as a free and unrestricted precompiled binary under the Apache 2.0 license from mujoco.org.

>MuJoCo's source code will be released through this GitHub repository once it is ready. In the meantime, the repository hosts MuJoCo's documentation

Why can't you just put it up now?  
In case you have any thoughts I'll tell you what I want it for... My goal is to replace the torque and the output code with a neural network trained on my real life, physical robot. A neural network trained to predict the state depending on the torque and last state.  
So, while I'm on the topic, any general advice how to proceed? Maybe where to begin? Like a more bite sized project before I tackle this?

cadop, 2021-11-22:
@MotorCityCobra  I think this is it https://github.com/openai/mujoco-py

MotorCityCobra, 2021-11-22:
Interesting. I knew about MuJoCo-py, but not the other. 
Still, I want the source files in C++ for MuJoCo. These both look like they’re in Python. .

saran-t, 2021-11-23:
You can actually already do this without needing to touch MuJoCo's source code at all. If your goal is to provide your own actuation model (e.g. by training a neural network to model real motors from logged data), the best way to proceed is by using actuator callbacks. More specifically:
- Define a C function of type [`mjfAct`](`gainprm = "0"`) that outputs the torque for a given actuator ID.
- Assign your function to MuJoCo's global function pointer to [`mjcb_act_bias`](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjcb-act-bias)
- In your model, create a [`<general>` actuator](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-general) with `gainprm = "0"` and `biastype = 'user'`.

See [here](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=actuator%20callback#mjcb-act-bias) for more detail on how actuator bias, gain, and activation translate to torque.

We still commit to fully open sourcing MuJoCo next year as originally announced. There are a number of technical action items on our side that we need to complete before we can open source. In the meantime, we will continue to release new binaries under permissive license terms that tackle some of the issues that we've posted on GitHub.

 I'll go ahead and change the title of this issue to reflect the technical question in your OP. If there are other specific issues that you feel require source access, please continue to reach out to us via new GitHub issues or discussions. Often, MuJoCo's existing API already offer the feature you need, and we can point you in the right direction. If that's not the case, we can work with you to see what we can do to help.

MotorCityCobra, 2021-11-23:
> * outputs the torque for a given actuator ID.

Thank you for the reply. Very enlightening.  
I haven't had a chance to get too far with the steps you laid out, and I thank you for the spoon feeding. I need it. But I might be picturing something a bit more elaborate.  
I'm seeing "bias for specified actuator", but I can't set a bias for just single actuators. The current state of every sensor and input to every motor has to set the bias in real time. I want inputs from all data about the robot inputted to my trained model inside the simulation. I want this trained model to set the bias.  
When training on the physical robot the neural net will have an input of every sensor on the robot. Accelerometers, gyros, and encoders. Other inputs to the neural net will be the torque, or torque called for into every actuator. Final inputs will be voltage, or something to factor battery power.  

The bias for every actuator will depend on the state of every sensor and the action of every motor and this will vary at every time step. What's the word? Dynamic, wholistic?
In my vision no simulator estimates the next state of a limb without inputting all current actions and positions. The actuators will have specific biases but they will always be in flux because I want the simulator sending all actions and states into my trained model in order to get the output for the prediction of where that force put the limbs.   

I repeated myself many times, but redundancy goes hand in hand with being precise.  

The model architecture:
Inputs will be the current position and current demands made for force from actuators.  
Outputs will be the positions of every limb.

MotorCityCobra, 2021-12-01:
@saran-t 

Here's what I have so far.

Modified from the humanoid.xml sample model.

    <actuator>
        <motor name="abdomen_y"       gear="200" joint="abdomen_y" />
        <general gaintype="user" gear="100" biastype="user" joint="abdomen_z" name="abdomen_z"/>
        <motor name="abdomen_x"       gear="200" joint="abdomen_x" />  

Create a my callback function  

    mjtNum my_callback(const mjModel *m, const mjData *d, int ssid)
    {
        return 10000;
    }

Inside the main function after the window is rendered.

        if (m)
        {
            int ssid = m->name_actuatoradr[1];

            mjcb_act_bias = my_callback;

            m->actuator_biasprm[1] = 2000;

I've tried changing a few things, but the model doesn't seem to be affected.

kevinzakka, 2023-11-21:
Closing with the introducing of [plugins](https://mujoco.readthedocs.io/en/latest/programming/extension.html#engine-plugins).

closed 2023-11-21 by kevinzakka