google-deepmind/mujoco #78: How do you compile in Windows?

MotorCityCobra, 2021-11-28:
In VS Code I'm using what looks like the [official extension](https://devblogs.microsoft.com/cppblog/now-announcing-makefile-support-in-visual-studio-code/) for running `make`.

This is the Makefile. It is the out of the box Makefile from the Windows version downloaded from the new official site.  

```COMMON=/O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/

all:
	cl $(COMMON) testxml.cc                 ../bin/mujoco210nogl.lib
	cl $(COMMON) testspeed.cc               ../bin/mujoco210nogl.lib
	cl $(COMMON) compile.cc                 ../bin/mujoco210nogl.lib
	cl $(COMMON) derivative.cc /openmp      ../bin/mujoco210nogl.lib
	cl $(COMMON) basic.cc ../bin/glfw3.lib  ../bin/mujoco210.lib
	cl $(COMMON) record.cc ../bin/glfw3.lib ../bin/mujoco210.lib
	cl $(COMMON) simulate.cc ../include/uitools.c ../bin/glfw3.lib ../bin/mujoco210.lib
	del *.obj```

This is the error.  

```Building target "" with command: "make.exe -f "C:\Users\ooo\Documents\mujoco210\sample\Makefile""
cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ testxml.cc                 ../bin/mujoco210nogl.lib
process_begin: CreateProcess(NULL, cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ testxml.cc ../bin/mujoco210nogl.lib, ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [C:\Users\ooo\Documents\mujoco210\sample\Makefile:4: all] Error 2
Target  failed to build.
Building target "" failed.```

MotorCityCobra, 2021-12-02:
I am able to build on Windows now using the x64 Native Tools Command Prompt for VS 2019.  
Going to the sample dir and typing `make`.  
There was a time when the files would not compile so I created a directory called `obj` in the sample file, which made it run.  

@saran-t 
@yuvaltassa 
I know I ask a lot of questions, but do close and leave an answers when I figure it out. I don't know if I'm muted or something for asking too many questions.  
My current problem is [What are the units is the `mjcb_act_dyn` callback using?](https://github.com/deepmind/mujoco/issues/83), if you want to through me a bone.

closed 2021-12-02 by MotorCityCobra

rohit-kumar-j, 2022-03-02:
@MotorCityCobra 
I had this error:
```
C:\mujoco-2.1.1\sample>make
g++ -O2 -I../include -L../bin -std=c++11 -mavx -pthread -Wl,-rpath,'$ORIGIN' testxml.cc    -lmujoco210nogl                             -o ../bin/testxml
C:/Program Files/Webots/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lmujoco210nogl
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:4: all] Error 1
```
Any suggestions on how to solve this?