google-deepmind/mujoco #83: What are the units is the `mjcb_act_dyn` callback using?

MotorCityCobra, 2021-12-01:
I've managed to get the mjcb_act_dyn to cause the model to move a specific joint to and fro from inside the `simulate.cc` file, which was the goal. But I'm not sure what units it is using, or even what kind of force it is exactly. How does this `dyntype` control compare to the slider control in the simulation? Which is between -0.4 and 0.4.

mjtNum dyn_callback(const mjModel *m, const mjData *d, int ssid)
{
    return dyn_factor;
}

        if (m)
        {
            int ssid = 1;

           // Inside the main function (for now)
            mjcb_act_dyn = dyn_callback;

            if ((int)((int)d->time % (int)4) == (int)0 || (int)((int)d->time % (int)4) == (int)1)
            {
                dyn_factor = -20;
            } else
            {
                 dyn_factor = 20;
            }
        }

Operative line in the xml file `<general dyntype="user" gaintype="user" biastype="user" gear="111" joint="abdomen_z" name="abdomen_z"/>`

I'm trying to get the units and forces down so I can compare them to actual motors and encoders in order to simulate.

erez-tom, 2021-12-03:
I believe the answer is covered by the relevant sections in the docs about the [actuation model](https://mujoco.readthedocs.io/en/latest/computation.html?#actuation-model) and the [`gear`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-general) attribute.
But perhaps I'm misunderstanding you?  I'd be happy to revisit the issue with a more specific question, if you're still unclear after reading the two sections (and the relevant background material).

closed 2021-12-03 by erez-tom

MotorCityCobra, 2021-12-03:
@erez-tom 
I see I'm going to have to get a better handle on more math.

My question...  
When I set `mjcb_act_dyn` to 50 it applies a force and the joint moves. When I set `mjcb_act_dyn` to -100 the joint moves. When I set `mjcb_act_dyn` to 300 the joint moves. The 50, -100, and 300 values are units of something. Maybe it's not a valid question, but I don't understand... 50 what?  
The documentation says "wi is computed by the user-defined callback mjcb_act_dyn. " and "We denote the vector of actuator activations w".
Maybe I was expecting something like foot pounds or horses.

I expected it to be some kind of force related to the slider actuator controls in the simulate sample file. They're usually between -0.4 and 0.4. I haven't seen what these controls are in the docs. I'm sure it's something to do with force applied.  

I'm getting the sense it's a rookie question that might be obvious to roboticists.