google-deepmind/mujoco #84: How to actuate a Cable Driven Parallel Robot (CDPR)

fenollp, 2021-12-01:
Hi, I'm simulating a CDPR consisting of 4 poles with 8 motors (top and bottom of each pole) linked via tendons to a rectangular lens in the center. The goal is to be able to command the motors so the lens can be oriented and translated.

I experimented with ball and hinge joints either on each corner of the lens or at its center but this seems to fix the lens in space.
From this [example code](https://github.com/deepmind/mujoco/discussions/74#discussioncomment-1695022) and what I could gather from the docs it seems two tendons require some kind of joint in between two geoms. But joints seem to remain in a given position while my lens position depends on these 8 tendons...

From here I'm unsure that what I want to achieve can fit in MuJoCo's modeling. Can you help explain the trick I should pull here?

<details>
<summary>cdpr.xml</summary>

```xml
<?xml version="1.0" encoding="UTF-8"?>
<mujoco model="cdpr">
    <compiler coordinate="global"/>
    <!-- <include file="scene.xml"/> -->
    <default>
        <default class="pole">
            <geom type="cylinder" size="0.02" rgba="1 1 1 1"/>
            <site type="sphere" rgba=".9 .9 .9 1" size="0.005"/>
        </default>
    </default>
    <asset>
        <mesh name="lightcone" file="lightcone.stl"/>
    </asset>

    <worldbody>
        <light pos="0 1 1" dir="0 -1 -1" diffuse="1 1 1"/>

        <!-- 3 by 3 square, 2 tall -->
        <body name="frame">
            <geom class="pole" name="pole_west"  fromto="-1.5 0 0  -1.5 0 2"/>
            <site name="Wt" pos="-1.5 0 2"/>
            <site name="Wb" pos="-1.5 0 0"/>
            <!-- <joint name="W" pos="-0.525 +0.7 2" type="ball" limited="true"/> -->

            <geom class="pole" name="pole_east"  fromto="+1.5 0 0  +1.5 0 2"/>
            <site name="Et" pos="+1.5 0 2"/>
            <site name="Eb" pos="+1.5 0 0"/>
            <!-- <joint name="E" pos="+0.525 -0.7 2" type="ball" limited="true"/> -->

            <geom class="pole" name="pole_south" fromto="0 -1.5 0  0 -1.5 2"/>
            <site name="St" pos="0 -1.5 2"/>
            <site name="Sb" pos="0 -1.5 0"/>
            <!-- <joint name="S" pos="-0.525 -0.7 2" type="ball" limited="true"/> -->

            <geom class="pole" name="pole_north" fromto="0 +1.5 0  0 +1.5 2"/>
            <site name="Nt" pos="0 +1.5 2"/>
            <site name="Nb" pos="0 +1.5 0"/>
            <!-- <joint name="N" pos="+0.525 +0.7 2" type="ball" limited="true"/> -->
        </body>

        <body name="lens">
            <freejoint name="root"/>
            <!-- <joint pos="0 0 2" type="ball" limited="true"/> -->

            <!-- 1.4/2=0.7 1.05/2=0.525 meters -->
            <!-- mass="3.5" kg -->
            <geom mass="3.5" name="lens" type="box" size="0.525 0.7 0.002" rgba="0 0 1 0.2" pos="0 0 2"/>
            <site name="W" pos="-0.525 +0.7 2"/>
            <site name="E" pos="+0.525 -0.7 2"/>
            <site name="S" pos="-0.525 -0.7 2"/>
            <site name="N" pos="+0.525 +0.7 2"/>

            <!-- distance from ground:2 - cone height:1.2 = 0.8 -->
            <geom mass="0" name="lightrays" rgba="1 1 0 0.1" type="mesh" pos="0 0 0.8" mesh="lightcone"/>
        </body>
    </worldbody>

    <tendon>
        <spatial name="Wt" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Wt"/>
            <!-- <geom geom="lens"/> is not sphere or cylinder -->
            <site site="W"/>
        </spatial>
        <spatial name="Wb" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Wb"/>
            <site site="W"/>
        </spatial>

        <spatial name="Et" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Et"/>
            <site site="E"/>
        </spatial>
        <spatial name="Eb" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Eb"/>
            <site site="E"/>
        </spatial>

        <spatial name="St" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="St"/>
            <site site="S"/>
        </spatial>
        <spatial name="Sb" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Sb"/>
            <site site="S"/>
        </spatial>

        <spatial name="Nt" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Nt"/>
            <site site="N"/>
        </spatial>
        <spatial name="Nb" width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 0.33">
            <site site="Nb"/>
            <site site="N"/>
        </spatial>
    </tendon>

    <actuator>
        <motor name="Wt" tendon="Wt" gear="10"/>
        <motor name="Wb" tendon="Wb" gear="10"/>
        <motor name="Et" tendon="Et" gear="10"/>
        <motor name="Eb" tendon="Eb" gear="10"/>
        <motor name="St" tendon="St" gear="10"/>
        <motor name="Sb" tendon="Sb" gear="10"/>
        <motor name="Nt" tendon="Nt" gear="10"/>
        <motor name="Nb" tendon="Nb" gear="10"/>
    </actuator>
</mujoco>
```

</details>

[lightcone.zip](https://github.com/deepmind/mujoco/files/7637686/lightcone.zip)

Second, side question: using the C/C++ or even some Python API, is there any way I could obtain the geometrical details of the ellipsis that is formed by the "lightcone" mesh intersecting with the ground plane? Could I set up a camera to get me this, looking up from below?

Thanks. Any info very much appreciated!

yuvaltassa, 2021-12-03:
Hi,

1. It is difficult to understand from your description what is the actual robot you are trying to simulate. If you could supply an image and/or CAD specification we might be able to help you model it.

2. No, there is currently no way of computing the intersecting polygon of two geoms in MuJoCo. Yes you could hack something with a camera but since your cone appears to be analytical, I would recommend just computing the intersection yourself using the standard equations for conic sections. It might be convenient to place an ellipsoid with the computed shape in the right location as a form of visual debugging.

closed 2021-12-03 by yuvaltassa

fenollp, 2021-12-03:
Sorry, I should have attached an image earlier. 
(You can find the related XML model in the previous post as well as the STL. The included scene.xml is from MuJoCo's models/)
[image]

This is the whole robot. 4 poles (cylinders) with each 2 tendons attached to each corner of the lens (blue transparent box). I added the yellow STL to represent light ray paths but this can be completely disregarded.

I'd like to add 8 motors (one at the top and bottom of each pole) that would pull/push on tendons (or cables) so the lens can be translated and rotated within the cube that the poles delimit. Each tendon should be in tension with ideal non-deformation nor slipping. This way each tendon is constrained by the 7 others.

How can I model this? Should I find a way to specify 8 ball joints between the poles and the lens? How can I make tendons behave like stiff cables?
Thanks

yuvaltassa, 2021-12-03:
Hi,

You can connect actuators to tendons, as described [here](https://mujoco.readthedocs.io/en/latest/computation.html#geactuation). If you use a `<position>` actuator (with high enough `kp`, but not too high), your actuator will try to enforce the length of the tendon (both pulling *and pushing* along the tendon, the latter being unrealistic for cables).

However, there is a deeper issue with your design. The lens has 6 DoFs while you intend to put 8 tendons there, which is too many constraints. Of course it is possible to move the lens with 8 tendons, but each primitive action will have to touch all your actuators, since you would be enforcing a 6D manifold in an 8D space. In fact, even with just 4 tendons (removing the bottom ones and letting gravity pull your object down), you can easily get conflicts. If you only use 3 tendons you should be able to freely move the lens in a 3D subspace. In order to get real 6DoF control with tendons you need to either compute the transformation yourself or redesign your mechanism.

I suggest you first see that you can connect actuators to tendons and understand how that works. Once you've figured this out, think about the constraint issue I described above and see how wish to proceed.

fenollp, 2021-12-03:
As shown in the provided XML I have already attempted to link actuators to tendons (as motors with gear >1). However when setting motor values (8 values, ranging -1;1, in the Controls pane to the right of the UI) no movement resulted. (I used a freejoint as I cannot understand how else I can arrange joints with my geoms and tendons).

I didn't know about `<position>` so I'll be looking into this. Ah yes I said pulling/pushing when talking about cables.. I meant in my specific assembly where a geom is pulled by cables in two opposite directions, movement in one direction translates to one motor pulling on a cable while the other opposite motor "let's go" of some length of the other cable. I wrote too hastily sorry.

Side question: are there any plans to add more examples to the docs? (that'd be great, thanks)
A link to an extensive open library of models would also be a great resource.

The CDPR I intend to simulate indeed has 8 cables for 6 DoF meaning it is over-constrained. This is by design. Here's [some literature](https://www.researchgate.net/figure/ReelAx8-un-manipulateur-a-8-cables-et-6-ddl-en-conguration-pleinement-contrainte_fig1_281184541) about [such robots](https://tel.archives-ouvertes.fr/tel-01516606/file/2015NICE4018.pdf).
Any one translation or rotation should require movement from all 8 actuators yes. Sorry again for not making this clear from the beginning.
Am I right in assuming MuJoCo can simulate such an over constrained cable robot?

Looking at my XML, do you have a suggestion as to a functional choice of joint.s?
Thanks

yuvaltassa, 2021-12-04:
Hi,

I loaded your model. The issue is exactly what I told you, the tendons are fighting against each other. The constraint solver is applying large forces to satisfy the length limits you have imposed (`<spatial ... limited="true" range="0 0.33">`) and your actuators can only apply 10 Newtons (`gear="10"`) which is not doing much. If you set `gear="100000"` the lens will move. However as I said, if you need to know the mapping between controls and lens pose, you will need to think more carefully about the geometry.

I recommend removing the explicit tendon length constraint, and working solely with high-kp `<position>` actuators.