google-deepmind/mujoco #86: Why is `mjData->qpos' just one number?

MotorCityCobra, 2021-12-05:
I would have assumed d->qpos would maybe give me an array of the x, y, z, position but it's just one number.  
If I had to guess the position is the height (y axis) from the ground?  

The docs say     `"mjtNum*   qpos; // position (nq x 1)"`.  
I looked but I didn't see anything like x, y, z position data.  
I would also like to have the angle of an actuator.

        if (m)
        {
            // qpos of joint #1
            std::cout << d->qpos[1];
        }

MotorCityCobra, 2021-12-06:
I'm delving into the sensors portion of the docs and it probably has all the position data I need.

    <sensor>
        <framexaxis name="lwaist" objname="lwaist" objtype="geom" />
        <frameyaxis name="left_larm" objname="left_larm" objtype="geom" />
    </sensor>

        if (m)
        {
            std::cout << d->sensordata[0][0] << " ";
        }

closed 2021-12-06 by MotorCityCobra

bycn, 2022-09-14:
To complete this question — they are not the height above the ground; they are the angles in *joint space*. Yuval's answer here: https://stackoverflow.com/questions/73296464/mujoco-what-is-the-difference-between-pos-qpos-and-xpos