google-deepmind/mujoco #88: How can I draw the trajectory of a robot's end-effector in mujoco viewer ?

haoqiangyang1998, 2021-12-12:
For example, I want to watch the trajectory of the end-effector in the `FetchReach-v1` environment during an episode. How can I draw the trajectory in mujoco's popup window?

yuvaltassa, 2021-12-13:
Are you using C/C++ or Python?
Before calling render, you need to create visual geoms using e.g. [mjvMakeConnector](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjv-makeconnector) and then add them to the array of visual geoms in [mjvSceme.geoms](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjvscene).

closed 2021-12-13 by yuvaltassa

lgkimjy, 2024-05-02:
@yuvaltassa I am currently drawing the trajectories of the end-effectors by using 

```
mjv_initGeom(test_geom, mjGEOM_SPHERE, size, NULL, NULL, color);
mjv_connector(test_geom, mjGEOM_LINE, 2, traj[i-1], traj[I]);
```

and the results of trajectories are shown as follows:
[image]

what I am curious is, can I assign ```group``` using cpp or in ```simulate.cc```? the reason why I want to do this is that I want to use the  ```group enable``` -> ```geom enable``` buttons to enable/disable (to show/ not to show) the trajectory that I am drawing. (because sometimes too many trajectories might confuse)

yuvaltassa, 2024-05-02:
No. Groups are used when filling the mjvScene (in mjv_updateScene) to decide what goes in it. I.e. by definition everything in the scene is visible and everything that is not visible is not in the scene. Does this make sense?

gywhitel, 2025-07-03:
@lgkimjy  Hi, I'd also like to draw the trajectory of robot base in mujoco, similar to your screenshot. 
I am confused of  creating`mjvGeom`. Do you mind sharing your context code?

yuvaltassa, 2025-07-03:
@lgkimjy look in our introductory tutorial, there is a section that does that

gywhitel, 2025-07-04:
Thanks, I find it out, and the codes posted below. I was confused by mjvScene.ngeom.
```cpp
void Simulate::addBaseTrajectory() {
  if (scn.ngeom - visual_traj_size_ < 0) {
    scn.ngeom += visual_traj_size_;
  }

  base_trajectory_.push_back(std::array<mjtNum,3> {d_->qpos[0], d_->qpos[1], d_->qpos[2]});
  if (base_trajectory_.size() > visual_traj_size_) {
    base_trajectory_.erase(base_trajectory_.begin());
  }
  mjtNum size[3] = {0.02, 0.02, 0.02};
  float rgba[4] = {0.9, 0.1, 0.1, 1};
  mjtNum mat[9] = { 1, 0, 0,
    0, 1, 0,
    0, 0, 1};
    
  for (size_t i = 0; i < base_trajectory_.size(); ++i) {
    mjtNum base_pos[3]= {base_trajectory_[i][0], base_trajectory_[i][1], base_trajectory_[i][2]};
    mjvGeom point;
    mjv_initGeom(&point, mjGEOM_SPHERE, size, base_pos, mat, rgba);
    point.category = mjCAT_DECOR;
    if (i > 0) {
      mjtNum base_pos_last[3] = {base_trajectory_[i-1][0], base_trajectory_[i-1][1], base_trajectory_[i-1][2]};
      mjv_connector(&point, mjGEOM_LINE, 2, base_pos_last, base_pos);
    }
    // Add geoms to mjvScene
    scn.geoms[scn.ngeom - visual_traj_size_ + i] = point;
  }    
  // printf("scene.ngeom: %d\n", scn.ngeom);
}
```