google-deepmind/mujoco #93: How do we solve this error on Apple Silicon?

matte-esse, 2021-12-16:
```
You appear to be missing MuJoCo.  We expected to find the file here: /Users/matteosacco/.mujoco/mujoco210

This package only provides python bindings, the library must be installed separately.

Please follow the instructions on the README to install MuJoCo

    https://github.com/openai/mujoco-py#install-mujoco
```

Essentially, How do we install on apple silicon?

wookayin, 2021-12-16:
It needs a non-trivial update to support the new binary. Please refer to https://github.com/deepmind/mujoco/issues/2#issuecomment-996018559

saran-t, 2021-12-16:
This issue is about mujoco-py, which isn't a project that we own or maintain. Having said that, since MuJoCo 2.1.1 doesn't change any public API compared to 2.1.0 we expect existing bindings to continue to work modulo changes in file names and location.

For dm_control, we will very shortly be pushing a small change that fixes the load path for MuJoCo and provide updated installation instructions. For mujoco-py, if the maintainers would like to update their code we'd be more than happy to work with them on it. In the meantime, I suggest that you mount the DMG for 2.1.1 then try the following:

```shell
mkdir -p ~/.mujoco/mujoco210/bin
cp /Volumes/MuJoCo/MuJoCo.framework/Versions/Current/libmujoco.2.1.1.dylib ~/.mujoco/mujoco210/bin/libmujoco210.dylib
cp -r /Volumes/MuJoCo/MuJoCo.framework/Versions/Current/Headers ~/.mujoco/mujoco210/include
```

Let us know if you get anywhere with this!

wookayin, 2021-12-16:
@saran-t I appreciate your efforts on the new MuJoCo release. I tried a similar workaround like yours, but it appears that a bit more steps are required in addition to that in order to make mujoco-py work. In short, `glfw` needs to be installed separately and the mujoco library needs to be put some `lib` directory that can be recognized (like `/usr/local/lib`).

@matte-esse Since this is about mujoco-py rather than mujoco itself, I wrote how I did in their repository rather than here. Here's the link: https://github.com/openai/mujoco-py/issues/662#issuecomment-996081734

saran-t, 2021-12-16:
This is our proposed installation scheme for `dm_control`: https://github.com/deepmind/dm_control/commit/84fc2faa95ca2b354f3274bb3f3e0d29df7fb337

Note that this is likely a somewhat temporary solution while we work out proper installation setup (we're considering msi for Windows, deb/rpm for Linux, and possibly pkg for macOS).

saran-t, 2021-12-16:
Let's close this issue and merge discussions into #2.

closed 2021-12-16 by saran-t