google-deepmind/mujoco #95: Does Mujoco have plans to support Apple Metal?

awkr, 2021-12-21:
Hi, since Apple releases M1 SoC which seems very powerful and efficient, does Mujoco have plans to support Apple Metal?

Coder-RG, 2021-12-21:
The devs are already working on it. Refer to #2

saran-t, 2021-12-21:
Do you mean for rendering or for compute?

awkr, 2021-12-22:
> Do you mean for rendering or for compute?

Sorry for not clarifying, features sorted by priority are:
1. compute: powered by MetalKit running on M1 and A14/15 chips
2. rendering

saran-t, 2022-03-18:
Sorry that it's taken a while to answer this one. I think at this point we can reasonably confidently say that Metal support isn't a priority for us, however we do intend to make it easier for people to plug MuJoCo into their own renderer. After open sourcing, if there is enough community interest in implementing parts of MuJoCo rendering in Metal we'd be supportive of the effort, but we probably won't be taking on the work ourselves.

closed 2022-03-18 by saran-t