google-deepmind/mujoco #96: joint limit reach (bug)

MyelinsheathXD, 2021-12-21:
Hey I just noticed an issue that  when joint limit violation, the joint will be frozen in that position. Then,  any external force can not  fix that limit violation.
In the video,  I illustrated the case.  

Both right and left legs knee joints frozen after  large external muscle force activation

https://youtu.be/GY49JBJ1n54

yuvaltassa, 2021-12-21:
Hi,

Thanks for the video! That's very helpful.

What you are seeing is not a bug. The feet are stuck behind the pelvis. You will be able to see this clearly if you hit 'F' and/or 'C' in `simulate`. To better understand what is happening, remember that MuJoCo has automatic parent/child [collision filtering](https://mujoco.readthedocs.io/en/latest/computation.html#coselection). So the lower leg and thigh do not collide (which allows the knee to bend), but the feet collide with the pelvis. So once your humanoid finds itself in the unfortunate position you show in the beginning of the video, there are two very strong opposing forces: The joint limit tries to straighten the knee, but the feet need to penetrate the pelvis for this to happen, so the contact is preventing it.

While of course it is possible to manually put the humanoid into this configuration, it should be very difficult to reach it with self-actuation.

I also recommend clicking on the various "disable" buttons in the "physics" tab. This will allow you to globally disable things like joint limits, contacts and gravity, which can be very helpful in diagnosing these situations.

closed 2021-12-21 by yuvaltassa

MyelinsheathXD, 2021-12-21:
thx for clarification!