google-deepmind/mujoco #101: How to control free joint with actuators?

tkreiman, 2021-12-31:
I am using mujoco_py to build a simulation for a hand. I have some some floating fingers that I want to control using actuators. How do I control a free joint that has 7 DOF (x,y,z, rotation quaternion) with acuators? From my understanding each actuator can control only 1 degree of freedom. My end goal is to have a position and orientation desired for each finger and move it there with the actuators.

Thanks for the help!

yuvaltassa, 2022-01-02:
There are multiple ways of doing this.

The simplest one is to use [mocap bodies](https://mujoco.readthedocs.io/en/latest/XMLreference.html#world-body-r). These are kinematic bodies whose position and orientation is set by writing into [`data.mocap_{pos,quat}`](https://github.com/deepmind/mujoco/blob/main/include/mjdata.h#L177) (rather than `data.ctrl`). Since these bodies have no velocities, it is usually recommended to put a weld between the mocap body and the actual free body you want to control, but you can try first without the weld. 

If this approach doesn't work for you, write back here and we will follow up.

NB the [composite example models](https://github.com/deepmind/mujoco/tree/main/model/composite) all have a mocap body that you can play with:

[image]

closed 2022-01-02 by yuvaltassa

tkreiman, 2022-01-03:
Thank you very much for the prompt response, this looks useful I have been playing around with it. Just to follow up quickly, when setting the mocap_pos the body teleports to this new position, as opposed to using actuators where the body would gradually move over to the desired location. Is there a way to mimic this continuous movement to a new position or do you recommend I just manually set the object's position interpolating between start and end position? To clarify I'm trying to avoid the body teleporting around the scene.

yuvaltassa, 2022-01-03:
That is correct, mocap bodies "teleport"; they have no velocity.

As indicated above, one way of reducing this unrealism is by having **two bodies**. The non-colliding mocap body is used as a control target, similar to the semantics of a position actuator, w.r.t a proper, colliding free body which is welded to it. The weld constraint applies a wrench to keep the free body attached to the mocap body, but the free body participates in the dynamics. Note that like all constraints, welds can be softened and otherwise tweaked with `solimp` and `solref`.

That said, interpolating is easy and I would probably try it first. Depending on your timestep, this might be a good solution. This is essentially what is happening in the GIF above, where the mouse-controlled mocap body is moving smoothly enough for the small jumps to not be an issue.

abhiyanthrik, 2024-06-15:
Hi, I am trying to run the examples for composite objects. I am using Ubuntu 22.04 and installed mujoco from [here](https://github.com/google-deepmind/mujoco/releases). Simulation is launching all right but I am not able to interact with the mocap. It does not move using the mouse or with any combination of keys with the mouse. 

Any suggestion or help?

yuvaltassa, 2024-06-16:
Double click to select, then move as usual (rotate:ctrl-left-drag, translate:ctrl-right-(shift)-drag)

abhiyanthrik, 2024-06-17:
I am not able to drag in z direction. I have tried looking for limit parameters or some constraints but didn't find any. Is there any way I can drag the mocap in z direction?

yuvaltassa, 2024-06-20:
Don't press shift.

The new video [here](https://mujoco.readthedocs.io/en/latest/programming/samples.html#sasimulate) might be helpful.

abhiyanthrik, 2024-06-20:
Got it, Thank you :)