google-deepmind/mujoco #102: MuJoCo.app (MacOS) crashes immediately after dragging and dropping model xml file

SupreethRao99, 2022-01-01:
Hello, 

I was trying to get up and running with MuJoCo but the app immediate crashed after I drag and drop any of the xml model files (eg humanoid100.xml) that come with the installation for MacOS. I'm running macOS Catalina (v10.15.7)

MacOS creates a crash report that looks like 
```
Process:               simulate [38051]
Path:                  /Applications/MuJoCo.app/Contents/MacOS/simulate
Identifier:            org.mujoco.mujoco
Version:               2.1.1 (2.1.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           simulate [38051]
User ID:               501

Date/Time:             2022-01-01 22:16:49.538 +0530
OS Version:            Mac OS X 10.15.7 (19H1519)
Report Version:        12
Anonymous UUID:        1F1A4272-A2C6-D421-1E89-51DA6773BA65

Sleep/Wake UUID:       E974EC3C-C505-445D-8A76-57D50533411C

Time Awake Since Boot: 300000 seconds
Time Since Wake:       2700 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Application Specific Signatures:
Graphics kernel error: 0xfffffffb
```

Is there anything that I'm doing wrong while running MuJoCo?

Thank You!

saran-t, 2022-01-01:
This seems to be the same issue as https://discussions.apple.com/thread/253405957

Can you see if there's a pending OS update for you?

SupreethRao99, 2022-01-02:
Hey @saran-t , I had a pending software update, after I installed the updated MuJoCo works perfectly fine. 
Thank You !

closed 2022-01-02 by SupreethRao99