google-deepmind/mujoco #104: how i can cancel the [Contact detection]

chinahuangyong, 2022-01-07:
how to cancle the contact detection in bodys.
[image]

lonelyfluency, 2022-01-07:
You can use `exclude` tag in your model xml file like this:

<contact>
        <exclude name="exclude_2_body_0" body1="base" body2="left_front_upper_leg"/>
       
</contact>

chinahuangyong, 2022-01-07:
good job！ ths.

chinahuangyong, 2022-01-07:
> You can use `exclude` tag in your model xml file like this:
> 
> ```
> <contact>
>         <exclude name="exclude_2_body_0" body1="base" body2="left_front_upper_leg"/>
>        
> </contact>
> ```

closed 2022-01-07 by chinahuangyong

yuvaltassa, 2022-01-07:
I'd like to add a relevant clarifying note.

MuJoCo by default excludes collisions between parent and child bodies, this is the reason there is no collision in the elbow joint in your model above. However, **this exclusion is not applied** if the parent is a _static_ body, i.e. the world body, or a dof-less descendant of the world. The reason is obvious: if those collisions were excluded, objects would fall through the floor etc. This collision filtering behaviour is documented [here](https://mujoco.readthedocs.io/en/latest/computation.html#collision-detection).

What this means is that if the root joint of a robot is removed, its top-level body _becomes static_ and starts colliding with its children. This implies that there is possibly a different solution to your problem: _add a free joint to your robot root_.

There is no reason not to use `<exclude>`, and the solution by @lonelyfluency is good one, but I thought this might help to clarify a common "gotcha" where people remove their robot's root joint and suddenly new contacts appear that were not there before.

chinahuangyong, 2022-01-18:
> I'd like to add a relevant clarifying note.
>
> MuJoCo by default excludes collisions between parent and child bodies, this is the reason there is no collision in the elbow joint in your model above. [...]

got it, think you.

livinamuk, 2022-04-07:
That was it, thank you.

LouisChen15, 2025-07-19:
> I'd like to add a relevant clarifying note.
>
> MuJoCo by default excludes collisions between parent and child bodies, this is the reason there is no collision in the elbow joint in your model above. [...]

Is there a way to disable the parent filter test? Since I want to let the parent and child object to collide with each other.

yuvaltassa, 2025-07-21:
https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-filterparent