google-deepmind/mujoco #110: Mujoco App or UI "Save xml" button file location

Olimoyo, 2022-01-13:
Hi, 

I am using the default macOS Mujoco App.
 
Is there a default location for where the XML files are stored when you press the "Save xml" button?
[image]

Thanks,
Oliver

Olimoyo, 2022-01-13:
I figured it out, it was under:
`/Applications/MuJoCo.app/Contents/Resources`

yuvaltassa, 2022-01-14:
Yes, this is indeed not great. Ideally, we would have a proper save dialog box.
If anyone wants to make a PR for this that would be appreciated.

saran-t, 2022-01-14:
Ideally we'd open up a "Save as..." dialog box when you're saving a file. Might be a bit platform-dependent though, but as @yuvaltassa said since the simulate code is open source we'd welcome any pull request for this.

I'll reopen the issue in case anyone wants to fix it.

saran-t, 2022-05-23:
A dialog box has been added [on macOS](https://github.com/deepmind/mujoco/blob/main/sample/macos_save.mm) so that the user can choose which directory they want the XML to be written into. 

We haven't added this for Windows or Linux yet. Doing this for Windows should be reasonably straightforward and not too dissimilar to what we did for macOS. Linux is a bit trickier since there isn't a "standard" API for this, and we'd prefer to not introduce a dependency just for a dialog box. If anyone has a good idea for this, please feel free to send us a PR.

closed 2022-05-23 by saran-t

csingh27, 2024-10-04:
Where is it stored by default in ubuntu?

rtnow, 2024-11-18:
> > Where is it stored by default in ubuntu?
> 
> In the directory where the simulate binary is run from.

what about windows？thanks for answer me