google-deepmind/mujoco #112: is it a bug to get body position

chinahuangyong, 2022-01-18:
Hello, I am use mujoco to simulate the quadruped mini cheetah,  mini cheetah is static, but i get the position from the sensor in mujoco, the position axis x is always Increase.

[image]

this is  mini cheetah xml file and stl file.
[mini_cheetah.zip](https://github.com/deepmind/mujoco/files/7885914/mini_cheetah.zip)

yuvaltassa, 2022-01-18:
Hi @chinahuangyong,

This is a bit hard to diagnose, in the model you attached the robot just falls over, so I can't reproduce. Did you remove the joints or add high-gain position actuators? If you send the model where this happens and we could follow up.

In the meantime I recommend turning on force visualisation (hit the 'F' key). For some setting you can get jittering forces which can cause this phenomenon. If that's the case perhaps try switching to elliptic friction cones (in the 'physics' tab) as the are completely isotropic, unlike pyramidal cones.

BTW, this is a nice model, can you point me to where you found it?

chinahuangyong, 2022-01-19:
yeah, i use pd control to control the joint position, the despite position;
```
RF:
x0.q[0] = -0.7;
  x0.q[1] = 1.;
  x0.q[2] = 2.715;
LF:
  x0.q[0] = 0.7;
  x0.q[1] = 1.;
  x0.q[2] = 2.715;

  x0.q[0] = -0.7;
  x0.q[1] = -1.0;
  x0.q[2] = -2.715;

  x0.q[0] = 0.7;
  x0.q[1] = -1.0;
  x0.q[2] = -2.715;
```
[image]

the model is convert urdf file to xml file, and add some constraint and others.

yuvaltassa, 2022-01-19:
Could you please make a model that shows this behaviour upon load in `simulate` without any additional code?
You can add position actuators using [`<position joint="joint_name" kp="gain"/>`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-position) (you probably want more damping in your joints btw), and you can change the semantics of your joints angles using [`<joint ref="refangle"/>`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-joint), such that the zero controls upon load will take the joints to `-refangle`

Thanks

saran-t, 2022-03-18:
Closing since there hasn't been a response for a while. Please reopen if required.

closed 2022-03-18 by saran-t