google-deepmind/mujoco #119: What are units of accelerometer and gyroscope?

markub3327, 2022-01-31:
Hello,

   in documentation is written about accelerometer and gyroscope, but nothing about their units.

Is gyroscope in deg/s or rad/s?
Is accelerometer in m/s2?

Thanks a lot.
Have a nice day.

yuvaltassa, 2022-01-31:
Gyroscopes always report rad / s.

Accelerometers report length-units / s^2. The actual length units are generally undefined, though they can often be inferred from the value of gravity under the assumption that you are on Earth. For the default (0, 0, -9.81), if you are on the surface of Earth, the unit of length can be inferred to be Metres. However if you are in free space with no gravity, the length unit is completely open to your own interpretation, as are units of force and mass etc, as long as they are all consistent. To phrase this with an example: the dynamics of a 1 metre spaceship that weighs 1 Kg and has a 1 Newton thruster are **exactly equivalent** to those of a 1 cm spaceship that weighs 1 gram and has a 1 Dyn thruster.

yuvaltassa, 2022-01-31:
That said, since this is a common question, we will add a section in the documentation to clarify.

Thanks for bringing this to our attention.

closed 2022-01-31 by yuvaltassa

yuvaltassa, 2022-03-17:
Following up, see [here](https://mujoco.readthedocs.io/en/latest/overview.html#units-are-undefined)

markub3327, 2022-03-17:
@yuvaltassa 

Thanks, it's very clear... 🙂

Thanks.