google-deepmind/mujoco #120: Importing models with meshes to Unity is slow

erez-tom, 2022-02-02:
When the Unity plug-in imports a model that includes STL files, the STL import process is a bottleneck.  A better approach would be to read the mesh directly from mjModel, but this is not yet implemented.

yuvaltassa, 2024-09-04:
@Balint-H is this still an issue, or can I close this?

Balint-H, 2024-09-04:
We still don't read the mesh from the MjModel, however the import and loading has heen significantly sped up by other changes to the C# components. I think the speed is no longer an problem, if that was the only reason for this issue, then we may close it. 

If for any use case mesh loading is still a bottleneck for someone, we can reopen the issue once its reported.

closed 2024-09-04 by yuvaltassa