google-deepmind/mujoco #127: ./record compressed with ffmpeg returns video of 0 second duration

SveaMeyer13, 2022-02-03:
Hi!

I'm new to MuJoCo and just tried to run the examples and got a bit stuck on the example for recording:

Following the [documentation](https://mujoco.readthedocs.io/en/latest/programming.html#sarecord) I first ran:
`./record ../model/humanoid/humanoid.xml 5 60 rgb.out`
and then 
`ffmpeg -f rawvideo -pixel_format rgb24 -video_size 800x800  -framerate 60 -i rgb.out -vf "vflip" video.mp4`

None of them gave me an error or a warning, rgb.out is of size 480 MB and video.mp4 of 250kB but it has a duration of 0 seconds.

I'd be grateful about any suggestions of what might have gone wrong :)

saran-t, 2022-03-18:
Hey, I missed this issue so sorry for the late response. Are you still interested in figuring this issue out? If you are, can you please give a bit more detail about your system?

kevinzakka, 2022-03-28:
I had the same issue and resolved it by adding `-vf "vflip,format=yuv420p"` .

So the full command is:

```bash
ffmpeg -f rawvideo -pixel_format rgb24 -video_size 800x800 -framerate 60 -i rgb.out -vf "vflip,format=yuv420p" video.mp4
```

kevinzakka, 2022-03-28:
@saran-t From searching on Google, it seems the rgb24 format [specifies](https://www.theimagingsource.com/support/documentation/ic-imaging-control-cpp/PixelformatRGB24.htm) that the organization of the pixels in the image buffer goes from left to right and bottom up. I think ffmpeg assumes this is the case and so reads the image upside down. So I think `record.cc` should be modified to store images upside down but I could be missing something...

closed 2022-05-19 by nimrod-gileadi

cirosantilli, 2024-01-15:
Neither the documented ffmpeg command nor kevinzakka's suggestion worked for me on Ubuntu 23.10 Lenovo ThinkPad p14s gen4 amd, the .mp4 output was garbled (though somewhat periodic, definitely just some line sze was wrong) and ffmpeg gave a warning:

```
[rawvideo @ 0x559ad068e340] Packet corrupt (stream = 0, dts = 1445).4735.5kbits/s speed=2.39x      
rgb.out: corrupt input packet in stream 0
[rawvideo @ 0x559ad0698400] Invalid buffer size, packet size 1459200 < expected frame_size 1920000
Error while decoding stream #0:0: Invalid argument
```

Also the file size is not divisible by 800 x 800:

```
2024-01-15_06-53-24@ciro@ciro-p14s$ stat rgb.out
  File: rgb.out
  Size: 2775859200      Blocks: 5421608    IO Block: 4096   regular file
Device: 252,1   Inode: 122951674   Links: 1
Access: (0664/-rw-rw-r--)  Uid: ( 1000/    ciro)   Gid: ( 1000/    ciro)
Access: 2024-01-15 06:52:07.777458016 +0000
Modify: 2024-01-15 06:53:24.050907637 +0000
Change: 2024-01-15 06:53:24.050907637 +0000
 Birth: 2024-01-14 00:30:17.013787880 +0000
~/git/mujoco/build
2024-01-15_06-53-26@ciro@ciro-p14s$ py
Python 3.11.6 (main, Oct  8 2023, 05:06:43) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 2775859200/(800*800)
4337.28
>>> 
```

lgkimjy, 2024-02-22:
@cirosantilli i have same issue, did you solve the problem? I am currently using Mac M1 Pro.

cirosantilli, 2024-02-22:
> @cirosantilli i have same issue, did you solve the problem? I am currently using Mac M1 Pro.

No. Someone needs to look into the bytes and try to figure out what they mean to reverse out an ffmpeg command.

kevinzakka, 2024-02-22:
The script should probably be modified to take in custom height and width arguments that way you don't have to do guesswork. That being said, I'm confused as to why the height and width are no longer 800 and need to investigate further.

kevinzakka, 2024-02-22:
Actually, the answer is simple. The humanoid XML was recently changed, specifically the [offscreen height and width](https://github.com/google-deepmind/mujoco/blob/7edacd39cdfeef72329975e0754fd48ca39b8c37/model/humanoid/humanoid.xml#L22) were increased from the previous value.

lgkimjy, 2024-02-22:
@kevinzakka Do you guys have any plan to add a record option in the simulation.cc?

yuvaltassa, 2024-02-22:
Adding record to MP4 from simulate is hard as it involves lots of dependencies. Saving raw video like record does will be easy. Would that suit your needs?