google-deepmind/mujoco #129: Unity warning: "Rename the assemblies to avoid hard to diagnose issues and crashes."

hubbardp, 2022-02-04:
At least with Unity 2021.2.8f1 on Windows, using the Package Manager to add the MuJoCo Unity package produces the following warning:

“Plugin 'Packages/org.mujoco/mujoco.dll' has the same filename as Assembly Definition File 'Packages/org.mujoco/Runtime/Mujoco.asmdef'. Rename the assemblies to avoid hard to diagnose issues and crashes.”

A solution that seems to be working (i.e., no more warning messages, both PlayMode and EditMode tests pass) is to rename `mujoco/unity/Runtime/Mujoco.asmdef` to `mujoco/unity/Runtime/Mujoco.Runtime.asmdef`.  This renaming is consistent with Unity's suggested naming conventions:

https://docs.unity3d.com/Manual/cus-asmdef.html

Note that in addition to renaming the `Mujoco.asmdef` file, the contents of the file must be changed slightly:
```
    "name": "Mujoco.Runtime",
```
And then references to it must be changed in the other `.asmdef` files.
`mujoco/unity/Editor/Mujoco.Editor.asmdef`:
```
    "references": [
        "Mujoco.Runtime"
    ],
```
`mujoco/unity/Tests/Editor/Mujoco.Editor.Tests.asmdef`:
```
    "references": [
        "UnityEngine.TestRunner",
        "UnityEditor.TestRunner",
        "Mujoco.Runtime",
        "Mujoco.Editor"
    ],
```
`mujoco/unity/Tests/Runtime/Mujoco.Tests.asmdef`:
```
    "references": [
        "UnityEngine.TestRunner",
        "Mujoco.Runtime"
    ],
```

Note that the name of the last such file, `mujoco/unity/Tests/Runtime/Mujoco.Tests.asmdef`, would be better and more consistent were it changed to `Mujoco.Runtime.Tests.asmdef`, but such a change is not strictly necessary to fix the warnings.

erez-tom, 2022-02-04:
This is very useful, thanks.  I'll prepare this change and push it in the next batch.

Anashb2007, 2023-10-16:
pls help me 
Plugin 'Assets/GoogleMobileAds/GoogleMobileAds.dll' has the same filename as Assembly Definition File 'Assets/GoogleMobileAds/GoogleMobileAds.asmdef'. Rename the assemblies to avoid hard to diagnose issues and crashes.

kevinzakka, 2024-08-30:
Going to close due to inactivity. Please re-open if this is still an issue.

closed 2024-08-30 by kevinzakka