google-deepmind/mujoco #130: Flag example in Unity: "PhysicsRuntimeException: BADQACC: NaN/inf in qacc"

hubbardp, 2022-02-04:
Using MuJoCo release 2.1.1 in Unity (2021.2.8f1) on Windows 10 Enterprise (version 10.0.19042 Build 19042) on an Alienware m15 laptop, the following example produces runtime exceptions:

https://github.com/deepmind/mujoco/blob/main/model/flag/flag.xml

I clone this repo, and apply the fix from issue 129 (https://github.com/deepmind/mujoco/issues/129), and import the MuJoCo package into Unity with the Package Manager.
I use the Unity editor's "Assets/Import MuJoCo Scene" menu item to load the `flag.xml` file.  There are no apparent errors during the importing.
I press Unity's `Play` button, and instead of seeing animation I see that the Unity editor console displays a steady stream of error messages like the following:

```
PhysicsRuntimeException: BADQACC: NaN/inf in qacc.
Mujoco.MjScene.CheckForPhysicsException () (at <path-omitted>/mujoco/unity/Runtime/Components/MjScene.cs:360)
Mujoco.MjScene.StepScene () (at <path-omitted>/mujoco/unity/Runtime/Components/MjScene.cs:319)
Mujoco.MjScene.FixedUpdate () (at <path-omitted>/mujoco/unity/Runtime/Components/MjScene.cs:88)
```

On the other hand, importing https://github.com/deepmind/mujoco/blob/main/model/tendon_arm/arm26.xml does work.  I get no errors during the import (only a warning about "MuJoCo's timestep (0.005s) ignored, please set fixedDeltaTime manually" and "The importer does not yet support tags <light>"), and pressing `Play` produces some animation and no errors in the console.

yuvaltassa, 2022-02-04:
This is due to our decision, documented [here](https://mujoco.readthedocs.io/en/latest/unity.html#design-principles), to use Unity's `Fixed Timestep` to override the timestep defined in the MuJoCo model. Since Unity's default time step is much bigger than the one in the MuJoCo model, the physics diverges. Copy the timestep from the model into Unity and it should work.

As a matter of user experience, from a Unity user's perspective, do you think we should reconsider that decision? If yes, what would you expect, that the Unity timestep is just ignored? That it is set to be MuJoCo's timestep as given in the model? We are happy to hear your opinions.

closed 2022-02-04 by yuvaltassa

erez-tom, 2022-02-04:
The warning about timestep mismatch didn't come up because `flag.xml` doesn't explicitly specify a timestep, instead relying on MuJoCo's default which is [2ms](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option) (in contrast to Unity's, which is 20ms).  We will rephrase the warning and error message, but if this will be an endemic issue to other users, we may opt to overwrite Unity's timestep at model import.

andreipit, 2022-03-10:
I was expecting to see a message in the selected gameobject's inspector, like: "The fixed time step in this model is different". And the button "Apply from the model".
"Overwrite Unity's timestep at model import" -  not transparent enough for me.

j2l, 2023-04-30:
After having this error, I changed the time step to 0.002 into project settings but still get:
```
PhysicsRuntimeException: BADQACC: NaN/inf in qacc.
Mujoco.MjScene.CheckForPhysicsException () (at C:/Users/admin/Downloads/mujoco-main/unity/Runtime/Components/MjScene.cs:380)
Mujoco.MjScene.StepScene () (at C:/Users/admin/Downloads/mujoco-main/unity/Runtime/Components/MjScene.cs:339)
Mujoco.MjScene.FixedUpdate () (at C:/Users/admin/Downloads/mujoco-main/unity/Runtime/Components/MjScene.cs:98)
```

erez-tom, 2023-05-02:
What happens if you try a ridiculously small timestep (e.g., 1e-5)?  Things will clearly be unusably slow, but you should expect to get no such errors.  You can then run several simulation while progressively increasing the timestep and find the critical value above which failures occur.  You can then visually inspect the simulation right at the critical timestep value, and hopefully see what goes wrong.