google-deepmind/mujoco #131: mj_saveLastXML uses locale's decimal separator on Linux

JeanElsner, 2022-02-05:
If your locale's setting on Linux uses a comma as a decimal separator (e.g. German locale),
`mj_saveLastXML` will write all numerical values using that separator. XML files saved that way
can then no longer be loaded correctly (`mj_loadXML` expects a dot separator).

Specifically, this breaks the Unity Plug-In's importer as `MjImporterWithAssets.ImportFile` first safes
the model in an XML file and then reads a string from that file to load the model again.

Changing your locale to use a dot separator fixes this of course, but this may not be a global setting that you want to use.

yuvaltassa, 2022-02-05:
Wow, that's a great bug report!
We would never have caught that ourselves, thanks. 
I expect we'll be able to fix this in the next release.

JeanElsner, 2022-03-18:
Great, I can confirm this issue is fixed now.

closed 2022-03-18 by saran-t