google-deepmind/mujoco #132: Site positioning problem

klonyyy, 2022-02-06:
Hello, 

I'm trying to draw ground reaction forces using site cylinders organized in a body like so:

XML file:
```
<default class='force_marker_size'>
    <site size="0.005 0.1"/>
</default>
.
.
.

<body name="force_LF" pos="0 0 0"> 
    <site name="force_LF_x" type="cylinder" rgba="1 0 0 1" euler="0 1.57 0" class='force_marker_size'/> 
    <site name="force_LF_y" type="cylinder" rgba="0 1 0 1" euler="0 0 -1.57" class='force_marker_size'/>
    <site name="force_LF_z" type="cylinder" rgba="0 0 1 1" euler="0.0 0 0" class='force_marker_size'/>
</body>
```
C code:
```
    int id = mj_name2id(m, mjOBJ_SITE,"force_LF_x");
    m->site_size[3*id+1] = force(0)/20.0f;     //change size (length)
    m->site_pos[3*id+0] = force(0)/20.0f;     //reposition it in the x axis (along the cylinder) 
    
    id = mj_name2id(m, mjOBJ_SITE,"force_LF_y");
    m->site_size[3*id+1] = force(1)/20.0f;     //change size (length)
    m->site_pos[3*id+1] = force(1)/20.0f;    //reposition it in the y axis (along the cylinder) 

    id = mj_name2id(m, mjOBJ_SITE,"force_LF_z");
    m->site_size[3*id+1] = force(2)/20.0f;     //change size (length)
    m->site_pos[3*id+2] = force(2)/20.0f;    //reposition it in the z axis (along the cylinder) 
```

because of the fact I cannot change the cylinder length during runtime (like with the XML fromto property), I just scale it with m->site_size[] property. Also, the cylinders are resized automatically on both ends, which means, whenever I want to draw a longer cylinder, I also have to reposition it, by the same value it was resized. I have found a problem with repositioning sites that have the euler property missing or filled with zeros in the XML file. Basically, the force_LF_z cylinder can be resized by cannot be repositioned without applying rotation in the euler property in the XML file. So simply saying:

This does not work (repositioning with m->site_pos[] property doesn't work):
`<site name="force_LF_z" type="cylinder" rgba="0 0 1 1" euler="0.0 0 0" class='force_marker_size'/>`

whereas this works as expected:
`<site name="force_LF_z" type="cylinder" rgba="0 0 1 1" euler="0.000001 0 0" class='force_marker_size'/>`

You can see on the GIF below how the Z-axis component behaves. Basically, X and Y are correct, but the Z component can be only resized, thus it goes in both directions.

[image]

After setting the euler property rotation to a tiny rotation in the XML file the problem is gone. It's not a big issue for me, but maybe you'd like to fix it (in case its really a bug, not my misunderstanding of basic concepts :) )

Best Regards,
Piotr Wasilewski

yuvaltassa, 2022-02-06:
Hi,
Could you please take a step back and explain why are you not using MuJoCo's built in contact force visualisation?
In `simulate` the relevant keys are 'C', 'F' and 'P'.

All visualisation options are available in the GUI on the right and keyboard shortcuts can be seen by right-clicking in the GUI.

closed 2022-02-06 by yuvaltassa

klonyyy, 2022-02-06:
Hello,
sorry, I forgot to explain the reason behind it - I'm doing it to check if my calculated forces (from motor torques using the jacobian matrix) actually match the ones calculated by MuJoCo. I'm going to use proprioceptive force sensing and this is my way of validating the jacobian matrix and other related calculations. This is why I need to dynamically change the cylinder lengths, not in the XML file.

yuvaltassa, 2022-02-06:
You did not answer my question. 
I did not ask why you want to visualise forces.
I asked: why are you not using the built-in force visualisation?

klonyyy, 2022-02-06:
I'm actually using the MuJoCo generated forces, I just turned their visualization off in the GIF so you could see the cylinders better. I hope this is what you meant? 
This is how it looks with the force visualization (simulation.c F key equivalent)  
[image]

Is there a way to draw these forces of user-defined magnitude? I thought they are like a built-in non-editable feature.

yuvaltassa, 2022-02-06:
Yes. You can change the [width](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-scale), [length](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-map) and [colour](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-rgba) of MuJoCo's built in forces.

klonyyy, 2022-02-06:
Alright, but can I use the visual/map/force property to set an arbitrary magnitude that won't be affected by the MuJoCo itself? From the visual/map/force description I understand its only a scaling factor: 
"This attributes controls the visualization of both contact forces and perturbation forces. The length of the rendered force vector equals the force magnitude multiplied by the value of this attribute and divided by the mean body mass for the model (see [statistic](https://mujoco.readthedocs.io/en/latest/XMLreference.html#statistic) element below)."

I'd like to have a visual indicator of my own force quantity not affected by the simulation engine at all.

yuvaltassa, 2022-02-06:
Ah ok! You want to draw your own externally calculated things in the 3D scene... this was unclear.

This is done by directly adding visual geoms to the `mjvScene`. No need to touch the XML or `mjModel`, it is unrelated the MuJoCo physics.

See if you can figure out how to do it, write here if you get stuck, we'll help.

Apologies that the C API is not super well documented, but everything you need is exposed. Read [this](https://mujoco.readthedocs.io/en/latest/programming.html#visualization) chapter, the `mjvisualize.h` header, and the relevant `mjv_` functions.

klonyyy, 2022-02-07:
Yes, that's my intention, sorry if I was unclear, I probably got fixed on my idea of representing data.. 

Thank you for the suggestion, I tried it and it works as expected :) I had some initial struggle with determining the order of operations so if anyone was interested in doing the same thing here you have a small example (based on mr. Emo's comment [here](https://roboti.us/forum/index.php?threads/rendering-geoms.3460/#post-3963)) : 

This is the part of the code where I add a custom geom (an arrow):

`        mjv_updateScene(m, d, &opt, &pert, &cam, mjCAT_ALL, &scn);

        mjvGeom* geomtest = scn.geoms + scn.ngeom++;

        mjv_initGeom(geomtest, mjGEOM_ARROW, NULL,NULL, NULL, NULL);

        geomtest->size[0] = 0.05;
        geomtest->size[1] = 0.05;
        geomtest->size[2] = 0.5;
        geomtest->pos[0] = 1;
        geomtest->pos[1] = 1;
        geomtest->pos[2] = 0;

        mjr_render(viewport, &scn, &con);`

I'd be grateful if you could tell me if I'm doing it right. 

In the end, I'd like to ask you about the [mjv_addGeoms](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjv-addgeoms) function - I'm having a hard time guessing what it does based on the description (its only mentioned twice in the docs). How does it add a geom to the existing mjvscene?

Joaopmoliveira, 2022-06-05:
Hi, 
My team and I are trying to use mujoco to simulate a robotic system which has been taught through demonstration learning, and we would like to show the clients the variance of this gaussian process. 
We decided to do this by generating a custom mesh instead of a geometry predefined in mujoco. 
would this be achievable with the functionality you described above? I was trying to achieve this by adding buffer by hand to the mujoco context, i.e. Adding a skin and reallocating the VBO and the other buffers, but it seems like a very hacky way to do it.
If you have any sugestion it would be appreciated.

yuvaltassa, 2022-06-05:
Hi @Joaopmoliveira,

1. Yes, what you want would be achievable this way, if you can do it with primitives (what the OP is about). Primitives might be a very good idea, depending on the data visualised.
2. It is also possible to do it with meshes (rather than skins), but the flow is different. You would change the mesh shape in `mjModel.mesh_vert` ([definition](https://github.com/deepmind/mujoco/blob/64bc6d27b242d5c5683b4a03dd36f1844b910742/include/mujoco/mjmodel.h#L726)) and then call `mjr_uploadMesh` ([declaration](https://github.com/deepmind/mujoco/blob/5cc73c30bfdc34e7b0f612e2ba25a7cdfc4a19ba/include/mujoco/mujoco.h#L616), [implementation](https://github.com/deepmind/mujoco/blob/64bc6d27b242d5c5683b4a03dd36f1844b910742/src/render/render_context.c#L209)). The integers (e.g. number of vertices) should stay fixed at runtime, but you can change the vertices and alpha channel (`mjModel.{geom_rgba or material_rgba}`) at runtime.
3. For skins, which are the only thing that give you the nice spline smoothing, you are right, there is no convenient way to do it right now. We are missing an `mjr_uploadSkin` function (so you would edit the floating point arrays in `mjModel.skin_xxx` and call this function), though it would be straightforward to add. Sounds like maybe you've already written it :sweat_smile:? Feel free to submit a PR or a gist of what you've done, if you think it could be useful.

If you wish to follow up, please do it in a new issue for visibility. Cheers.

Krishnendu1984, 2022-08-28:
Kindly tell me how to get ground reaction forces and its location on the say for foot of robot

v-r-a, 2022-08-28:
@Krishnendu1984 
Relevant links:
1. [Contact data](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjcontact) for extracting contact information.
2. [Function](https://mujoco.readthedocs.io/en/latest/APIreference.html?highlight=mj_contactForce#mj-contactforce) to get the value of contact forces:
`void mj_contactForce(const mjModel* m, const mjData* d, int id, mjtNum[6] result)`
`id ` is the contact ID. 

Now you may want to know the bodies corresponding to this contact `id`.

You can get geometries in contact by checking: 
`int geom1_id = d->contact[id].geom1` and 
`int geom2_id = d->contact[id].geom2`

Now, to know the bodies to which these geometries belong:

`int body1_id = m->geom_bodyid[geom1_id]` and similarly for geom2.

You may further use [mj_id2name](https://mujoco.readthedocs.io/en/latest/APIreference.html#mj-id2name) to check whether it is the body you were looking for. (Pass `mjOBJ_BODY` in place of `int type`).

You can loop the above code for contact `id` from `0<= id < d->ncon` to search for the required pair of contact.

`d->ncon` gives the number of active contacts at the moment.
You may search "`geom_bodyid`" in [mjModel](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjmodel))
You may want to read more explanation on why to use mj_contactForce() [here](https://mujoco.readthedocs.io/en/latest/programming.html#contacts).

Krishnendu1984, 2022-08-29:
Ok thats the force but its location or coordinates how to get?