google-deepmind/mujoco #133: Unity import fails

armeausstahl, 2022-02-06:
I tried to import an xml file from the models folder in unity.

**Expected behaviour:** 
model should render correctly with textures.

**Actual behaviour:** 
Its not loaded correctly. Textures are missing and I got the following error in the console

NullReferenceException: Failed to create Mujoco runtime.
Mujoco.MjScene.StepScene () (at /mujoco-main/unity/Runtime/Components/MjScene.cs:325)
Mujoco.MjScene.FixedUpdate () (at /mujoco-main/unity/Runtime/Components/MjScene.cs:88)

erez-tom, 2022-02-06:
This error comes from the run loop, after hitting "Play", which I find confusing:
- Did any errors or warnings come up during import?  Those errors should be the relevant ones to inspect.
- Assuming import completed without errors, I would still expect other errors (e.g., from model creation / simulation init) to be displayed before FixedUpdate is stepped, and again those would be the more informative ones.
Do you have "Error Pause" on?  That helps noticing the first error, that's most likely to be actionable -- after the first error, the state of the system is undefined...

armeausstahl, 2022-02-09:
Hey @erez-tom 

Yeah I forgot to use the "Error Pause" option and figured out that it shows the following error:

IOException: Error loading the model: Error: sizes must be positive in object 'Sphere_2' (id = 0)
Object name = Sphere_2, id = 0, line = 1, column = -1

But I was wondering why that is happening. Because I just imported a basic model from the repo.

JeanElsner, 2022-02-09:
Do you happen to use Linux with German localisation (e.g. format/decimal separator)?

I got the same error when trying to import the sample models. This could be a duplicate of https://github.com/deepmind/mujoco/issues/131
If so, changing your system settings should fix the import for now.

saran-t, 2022-03-15:
Closing this as a potential duplicate of #131 but please feel free to reopen if we're wrong about that!

closed 2022-03-15 by saran-t