google-deepmind/mujoco #140: How to measure contact force and its direction

CarlDegio, 2022-02-13:
What is recommended way to measure contact force (both scale and direction)? I use mujoco to build a simulation of manipulator with end effector is a peg, but the measurement of force it's imposed is difficult.

plan A: use touch sensor, but is can't measure the force direction.
plan B: use force-torque sensor. In this way, I add a zero mass small sphere at the end of peg, and use force-torque sensor to measure the force between the peg and the sphere, which is contact force on sphere. Roughly successful. But things become strange since I contrast the force measured by force-torque sensor and touch, they have small difference in scale (about 10%. And because zero mass sphere, it makes no sense). I don't know which is correct.

So is there a better plan? or why plan B has problem.

closed 2022-02-13 by yuvaltassa

CarlDegio, 2022-02-13:
Yeah, I test it and find out when friction is zero the touch and force are the same.

About the option of touch, I think that is a great idea. Further, many colleagues get confused when use touch to measure contact force, and contact force appears more and more frequently in academic. So it will definitely be easier to use if touch sensor could return the force component to represent the direction of touch. But it may be difficult when it comes to multi-point contact situation.

Why zero mass is not recommended, for small single body (sphere at the end of cylinder), zero mass may not cause singular problem.

Krishnendu1984, 2022-09-02:
i have used touch sensor below foot it is showing exact value  122 each at all four corner but when i used force sensor it is showing all zero values .Kindly clarify

CarlDegio, 2022-09-03:
> i have used touch sensor below foot it is showing exact value 122 each at all four corner but when i used force sensor it is showing all zero values .Kindly clarify

force sensor measure the force between two body ( such as leg body and foot body, and contact appear in foot body). And touch sensor just measure the contact on foot body.