google-deepmind/mujoco #142: Unity MJHingeJoint no way to manage axis through the Unity Editor

makeplayhappy, 2022-02-15:
https://github.com/deepmind/mujoco/blob/175d25cd9fe9e821885655f67819811bcf633561/unity/Runtime/Components/MjHingeJoint.cs#L37

On a MjHingeJoint there is no way to modify the axis via the editor. Possible solution could be to set to Vector3.right and then handle the local rotation at runtime startup.

erez-tom, 2022-02-15:
By construction, the `MjHingeJoint` component is required to have its own game object, and you control the rotation axis by pointing the transform's X axis in the desired direction (in particular, via the Editor); what other way to modify the axis would you want?

makeplayhappy, 2022-02-15:
Ah, thats great, makes sense now. 
I'd assumed it would be defined in the same way as in the XML definitions like
`<joint axis="0 1 0" name="hinge" pos="0 0 0" type="hinge"/>`