google-deepmind/mujoco #145: Projecting points onto rendered field of view

ishank-juneja, 2022-02-16:
I have a state estimator that estimates state from images and I want to project back these estimates onto the rendered frames I get from mujoco (that I saved to disk). My state estimates are in the same coordinate frame as the ground-truth states obtained from mujoco (ground-truth state is how I trained my state-estimator) 

I am trying to programmatically project my state-estimates back onto to the rendered image observations that I have saved to disk. To do this I need a correspondence between the pixel coordinate frame and mujoco's coordinate frame.   

So far I have been using some numbers that I found manually via hit and trial to perform this translation but I wanted a more general way of doing this. Help would be really appreciated.

The pieces of code that I think are relevant (but am not sure how I can use to do what I want are)

Camera setup from my `.xml` file
```
<camera name="rgb" mode="fixed" pos="0 4 0.0" euler="-90 0 180"/>
``` 

Viewer setup code snippet from my `mujoco_py` model definition
```
class MyEnv(mujoco_env.MujocoEnv, utils.EzPickle):
    def __init__(self):
        xml_path = os.path.abspath(mypath)
        mujoco_env.MujocoEnv.__init__(self, xml_path, frame_skip=1)
        utils.EzPickle.__init__(self)
        # Must come after model init
        self.reset_model()

    def viewer_setup(self):
        v = self.viewer
        v.cam.trackbodyid = 0
        v.cam.distance = self.model.stat.extent * 0.5
        v.cam.lookat[2] = 0.12250000000000005 
```

The `self.model.stat.extent` attribute seemed like what I needed but when I printed out its value-
```
print(self.model.stat.extent)
>>1.3
```
I got about 1.3 which did not match the value I was seeing that was closer to 1.7 ...

yuvaltassa, 2022-02-16:
This is already [implemented](https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/engine.py#L706) in dm_control. Check out the cell in the [tutorial colab](https://github.com/deepmind/dm_control/blob/master/dm_control/mujoco/tutorial.ipynb) titled `Projecting from world to camera coordinates`.

closed 2022-02-16 by yuvaltassa

ishank-juneja, 2022-02-16:
Thank you for your reply.
 
Do mujoco_py and dm_control use the same wrappers to interact with the simulator? If so then maybe I could use the dm_control source code to replicate the process.

I ask because porting my model from mujoco_py (gym) interface to dm_control seemed like a really long process since although the dm_control documentation is pretty good, the mujoco_py docs are not very helpful and I haven't seen any examples of "translation" between these 2 interfaces that I could use as a template to do my translation.

ishank-juneja, 2022-02-17:
@yuvaltassa I know the issue is closed, but would really appreciate your response or any links in the right direction since I am a bit lost about this.

Thanks again!

ishank-juneja, 2022-02-18:
I see that the computation there is implemented from scratch in the tutorial.

Resolved.