google-deepmind/mujoco #147: I have a simple model made of two meshes with a joint connecting them. Actuator is not working.

MotorCityCobra, 2022-02-17:
My "model" is an egg slicer kitchen utensil made of two meshes and a hinge. I can't get the actuator to move the joint at all.  
As it is now the egg slicer hinge swings open 180 degrees when the model loads and I can't move it with the actuator's slider.

I got the actuator's x, y, z locations from the joint's location in SolidWorks. The coordinates are to the hundredth decimal. It looks like the actuator is in the right spot when I 'render' it in the simulation.  

I've tried every variation for `contype` and `conaffinity` in the `default class` of the XML file.  

I've seen Emo write in the old forum that meshes should have cones, cubes, and cylinders, etc. in place of meshes. The meshes should have no mass to the simulator? I'm not sure why, maybe I could get some clarification about this. Do meshes not work as objects with mass and hard surfaces? If not this really going to limit my plans for robots tasks I want to simulate.

You can download the XML and meshes [here](https://drive.google.com/drive/folders/1Fi7lo0S43jWJacjZj8xuRTxsYfwMXGWQ?usp=sharing)  

The XML  
```
<mujoco model="egg_slicer">
	<compiler angle="radian" meshdir="meshes/"/>
    <default class="viz">
        <geom contype="1" conaffinity="1" type="mesh" rgba=".51 .6 .7 1"/>
    </default>

	<asset>
		<mesh name="base"         file="egg_mash1.stl" />
		<mesh name="swing"         file="egg_mash2.stl"/>
    </asset>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/> -->
        <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    </asset>
    <worldbody>

        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>
        <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

        <!-- start building our model -->
        <body>
            <geom name="slicer_base" mesh="base" />
            <body pos="0 0 0">
                <joint name="hinge" pos="228.60 120.56 33.58"
                axis="0 0 1"
                limited="true" range="-3.0 0"  />
                <geom name="slicer_swing" mesh="swing" />
            </body>
        </body>

    </worldbody>

    <actuator>
        <motor name="hinge" gear="500" joint="hinge"/>
    </actuator>
</mujoco>

```

[image]

yuvaltassa, 2022-02-19:
Hi,

Your link doesn't work as it's not shared. Please update the link to use GitHub rather than Drive so linked files are permanent.

From visual inspection of your model `<joint pos="228.60 120.56 33.58"...` seems strange. I'm not sure if your distance units should read as metres or centimetres, but either way that seems off.

MotorCityCobra, 2022-02-19:
Sorry about the link. I updated it in the OP and [here](https://drive.google.com/drive/folders/1Fi7lo0S43jWJacjZj8xuRTxsYfwMXGWQ?usp=sharing) it is as well.    

That location of the hinge is correct. In the original stl files the zeros for x, y, and z were a little bit outside of the frames of the objects.
Since posting I have managed to see the hinge swing some times after adding `inertial` tags below the geoms. The OP question has changed to what goes in the inertial tags.  

Sorry for the rookie questions. I have looked around for an answer to these questions.
What is the "pos", the position, of the inertial?  
I'm not sure of the diaginertia either. I've copied from the original model in SolidWorks. The object, (the model) tends to flap around like a butterfly.  
I don't get far in physics, and I'm sure there is a good reason, but why doesn't MuJoCo simulate the motion and collisions of objects from mass and shape of the objects alone?
Also, I notice a lot of the models with meshes have cylinders, cubes, and other simple objects at the locations of the meshes to stand in for the collisions. Is there a problem with using meshes alone for simulations? I'd like to model real life objects in SolidWorks and bring them into MuJoCo so I can improve my robot's functioning. I'm afraid using cylinders and capsules instead of meshes might widen the simulation vs real life gap.

The worldbody part of the code now.  
```

    <worldbody>

        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>
        <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

        <!-- start building our model -->
        <body>
            <joint type="free" limited='false' damping="0" stiffness="1" armature="0"/>

            <geom name="slicer_base" mesh="base" class="viz"/>
            <inertial pos="0 0 0" mass="0.49" diaginertia="0.01 0.02 0.02"/>

            <body pos="0 0 0">
                <joint name="hinge" pos="228.60 120.56 33.58"
                axis="0 0 1"
                limited="true" range="-3.0 0" /> -->
                <geom name="slicer_swing" mesh="swing" class="viz"/>
                <inertial pos="0 0 0.01" mass="0.70" diaginertia="0.02 0.02 0.03"/>
            </body>
        </body>
    </worldbody>

```

The mass information from SolidWorks for the top piece.
[image]

yuvaltassa, 2022-02-20:
In `simulate`, hit the 'C' or 'F' to see that your two bodies are colliding. This is causing huge forces, which explain what you are seeing. To understand why they are colliding, read my comment [here](#104).

Also in `simulate`, hit the 'H' key to see the convex hulls of your bodies. This is the thing that is actually colliding, not your visual geoms.

closed 2022-02-20 by yuvaltassa

MotorCityCobra, 2022-02-20:
Thank you. I appreciate you replying even though I have a lot of unfocussed questions.  
Since the program is under new management I would like to put two cents in for new docs... Maybe you don't want the docs to pander people who haven't learned the basics of physics and inertia, but I would appreciate a tutorial on how to get an object to 'act' like it would in real life. 
I would like to learn on my own, but as I said, the docs say the "Position of the inertial frame" for `pos` in the `inertial` tag. I don't know how to search for this in Google results of educational materials.