google-deepmind/mujoco #150: Release the source code and build instructions

lapp0, 2022-02-19:
Hi, I'm trying to use mujoco, but cannot link `libOSMesa.so.6` without downgrading my operating system. It would be extremely helpful if we could build from source rather than relying on prebuilt binaries.

When is mujoco expected to become open source software?

lapp0, 2022-02-19:
In case anyone found this issue trying to resolve the `libOSMesa.so.6` problem, I solved it via `patchelf --replace-needed libOSMesa.so.6 libOSMesa.so.8 $src/lib/libglewegl.so`

yuvaltassa, 2022-02-20:
Please see our announcement [here](#126) regarding open sourcing,

closed 2022-02-20 by yuvaltassa

saran-t, 2022-02-20:
Sorry about the `libOSMesa.so.6` thing! I've only noticed it last week too. This is caused by us building our binaries using PyPA's `manylinux2010` Docker image in order to ensure that our binaries are compatible with as many Linux systems as possible.

Note that `libmujoco.so` itself only depends on glibc, and OSMesa is only required by `libglewosmesa.so` which isn't part of MuJoCo, but rather a build of [GLEW](http://glew.sourceforge.net/), which you can already build from source if required. We're only shipping GLEW binaries as part of our Linux bundle for convenience -- you don't actually need to use them.

That said, as you rightly pointed out, `libOSMesa.so.6` seems to be ABI-compatible with `libOSMesa.so.8`. Indeed if you `apt install libosmesa6-dev` on Debian or Ubuntu, `libOSMesa.so.6` is a symlink to `libOSMesa.so.8`.

In summary, your options around the OSMesa issue are:
- `patchelf --replace-needed libOSMesa.so.6 libOSMesa.so.8 libglewosmesa.so` (the `libglewegl.so` in your previous comment was a typo)
- `ln -s libOSMesa.so.8 libOSMesa.so.6` wherever `libOSMesa.so.8` is on your system
- Build GLEW from source

lapp0, 2022-02-21:
Thanks both of you.

Your link is broken, but I found the announcement: https://github.com/deepmind/mujoco/issues/126

Looking forward to the FOSS release!

saran-t, 2022-03-15:
Just a quick followup on the OSMesa DSO version. For our [2.1.2 release](https://github.com/deepmind/mujoco/releases/tag/2.1.2), the copy of `libglewosmesa.so` that comes bundled with MuJoCo has reverted to depending on `libOSMesa.so.8` as it was in previous versions.