google-deepmind/mujoco #151: Could I please get a better explaination of what "pos" is in the inertial tag within a body?

MotorCityCobra, 2022-02-21:
`pos` in the body tag is an x, y, z position in the model. The docs say it is the position of the inertial frame. I've been looking through the docs and many physics robot tutorials trying to figure out what this means and where to derive it for each body.

To quote the documentation "Position of the inertial frame. This attribute is required even when the inertial properties can be inferred from geoms. This is because the presence of the inertial element itself disabled the automatic inference mechanism."  

I don't think it is center of mass. Is it the position of the joint? I can pull of the mass of meshes in SolidWorks and get the inertial matrix. I believe these values go in the `diaginertia` field, and are not the position of the frame of reference.  

I will post an example with two meshes and one joint.  
I think the second joint (below the first) might have the correct `pos` for inertial. I set it as the position of the joint. Was that correct? I don't know what it should be for the first mesh. The joint didn't work and the center of mass didn't work. Zeros didn't work.  

```

<mujoco model="egg_slicer">
    <compiler angle="radian" meshdir="meshes/"/>
    <default class="viz">
        <geom contype="5" conaffinity="1" group="1" type="mesh" rgba=".51 .6 .7 1"/>
    </default>

    <asset>
        <mesh name="base"         file="egg_mash1.stl" />
        <mesh name="swing"         file="egg_mash2.stl"/>
    </asset>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/> -->
        <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    </asset>
    <worldbody>

        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>
        <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

        <!-- start building our model -->
        <body>
            <joint type="free" limited='false' damping="0" stiffness="1" armature="0"/>

            <geom name="slicer_base" mesh="base" class="viz"/>
                <inertial pos="0 0 0" mass="1" diaginertia="0.01 0.01 0.01" />

            <body pos="0 0 0">
                <joint name="hinge" pos="228.60 120.56 33.58"
                axis="0 0 1"
                limited="true" range="-3.0 0" /> -->
                <geom name="slicer_swing" mesh="swing" class="viz"/>
                <inertial pos="228.60 120.56 33.58" mass="703.29" diaginertia="1.36 3.96 5.21"/>
            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="hinge" gear="100" joint="hinge"/>
    </actuator>
</mujoco>

```

yuvaltassa, 2022-02-22:
The position of the inertial frame is indeed the the center of mass of the body. This is described [here](https://mujoco.readthedocs.io/en/latest/overview.html#kinematic-tree).

closed 2022-02-22 by yuvaltassa

MotorCityCobra, 2022-02-23:
I see!  

I finally got the center object (the model) to fall like a normal object like this would fall, though it is slow motion. 
My problem is the hinge joint will not open.  
When I had the wrong info in the `inertial` I could use the slider in the 'control' section to open and close the hinge.  

       ```
     <geom name="slicer_base" mesh="base" class="viz"/>

                <inertial pos="132.89 127.79 104.58" mass="60.80" diaginertia="85.62 308.99 247.64" />

            <body pos="0 0 0">

                <geom name="slicer_swing" mesh="swing" class="viz"/>
                <joint name="hinge" pos="228.60 120.56 104.58"
                axis="0 0 1"
                limited="true" range="-3.0 0" />
                <inertial pos="116.12 134.66 104.58" mass="87.19" diaginertia="172.34 641.78 491.05"/>
```