google-deepmind/mujoco #152: Joint between two mesh "model" only works under certain conditions

MotorCityCobra, 2022-02-23:
I have a model made of two meshes with a hinge joint connecting them that does not work. The actuator at this joint does not swing open or close when I move the slider.  
The joint does open if I comment out the "joint" in the main body that makes the model fall to the ground.  
The joint and the actuator will work if I have some combination of the incorrect `pos` and/ or the incorrect `diaginertia` information entered for the meshes. That is, I can move the slider to open the joint. But with the wrong inertial information the center of mass is off and the object, (the model) flaps around on the ground.  

[Here are the meshes](https://drive.google.com/drive/folders/1w3nh8ptOvkwdV8dEUAxSSDpWyGk40Fy7?usp=sharing) and the code below is the XML with the joint type "free" to make the object fall to the ground commented out. If you uncomment it it looks like the object falls to the ground naturally like it would in real life.
```

<mujoco model="egg_slicer">
    <compiler angle="radian" meshdir="meshes/"/>
    <default class="viz">
        <geom contype="1" conaffinity="1" group="1" type="mesh" rgba=".51 .6 .7 1"/>
    </default>

    <asset>
        <mesh name="base"         file="egg_mash1.stl" />
        <mesh name="swing"         file="egg_mash2.stl"/>
    </asset>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/> -->
        <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    </asset>
    <worldbody>

        <geom name="floor" pos="0 0 -5.5" size="0 0 1" type="plane" material="matplane"/>
        <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

        <!-- start building our model -->
        <body>
            <!-- <joint type="free" limited='false' damping="0" stiffness="1" armature="0"/> -->

            <geom name="slicer_base" mesh="base" class="viz"/>

                <inertial pos="132.89 127.79 104.58" mass="60.80" diaginertia="85.62 308.99 247.64" />

            <body pos="0 0 0">

                <geom name="slicer_swing" mesh="swing" class="viz"/>
                <joint name="hinge" pos="228.61 120.56 135.58"
                axis="0 0 1"
                limited="true" range="-3.0 0" />
                <inertial pos="116.12 134.66 104.58" mass="87.19" diaginertia="172.34 641.78 491.05"/>

            </body>
        </body>
    </worldbody>

    <actuator>
        <motor name="hinge" gear="1000" joint="hinge"/>
    </actuator>
</mujoco>
```  

The model with the center of mass rendered without the joint for the main body commented out

[image]

The model without the joint for the main body commented out
[image]

The model with the joint on the main body commented out
[image]

The meshes in SolidWorks that shows the xyz position of the hinge joint between the two meshes.  

[image]

yuvaltassa, 2022-02-23:
You are using inconsistent units. The length units you are copying from your CAD software is mm, the masses are in grams, the inertias are in I don't know what, yet you haven't changed the default gravity from the default -9.81. So there is no way for the units of "gear" to make sense. If I uncomment your free joint and set gear=100000 the actuator works fine, modulo the weird behaviour coming from the stiffness you put in the free joint (don't do that).

Use consistent units. I'd recommend either MKS or CGS, but in the latter case change gravity to (0, 0, -981).

closed 2022-02-23 by yuvaltassa

MotorCityCobra, 2022-02-24:
> inconsistent units

I don't know if maybe you want me to make another post about this, but I don't understand how to get into MKS.  
Does it have to do with how I export the model / stl files from SolidWorks? Do I need to change units in Solidworks?  
Are the meshes not the same size no matter what the units are?  
When I input the units as Meters, mass as Kilograms and diaginertia as meters cubed the center of mass is way off.
But I don't understand why, because the object should be the same size and mass.

I've tried exporting the meshes from SolidWorks with different units set. The things I can think to do next is scale the model, the meshes down 1000x in SolidWorks.  
I've asked before what units SolidWorks was in but the answer was that SolidWorks has no units.

yuvaltassa, 2022-02-24:
Meshes can be converted from mm to metres with `<mesh scale="1e-3 1e-3 1e-3"/>`
For masses just multiply by the right factor.
Inertias have units of mass*length^2.
Once you units are consistent your torques (actuator output) will be in some sensible unit like Nm or Dyn and you'll be able to reason about it.

MotorCityCobra, 2022-02-25:
I appreciate the answers. I know this level of help is above and beyond.
But I'm still more or less lost.  

When I scale the meshes down with `<mesh scale="1e-3 1e-3 1e-3"/>` I can barely see the model after zooming in as far as I can.
The actual object when folded is about 8". It's a little big for this object in real life. It's more like 5" (127 mm) in real life. My point is, the object brought into MuJoCo is a good size and when I use MuJoCo for simulations I would like to import objects in real size. I would like a real life object to 'act' like it would in real life at 0 0 -9.81 with the object's mass as is, which in this case is a hundred something grams whether its 8" or 5". In other words, the dims of the meshes and mass in the XML seem like they are what they are in real life.
Sorry, if this is too long winded. If it's 210 mm it's still 0.21 meters, so I'm still confused about what needs to change to have a sensible torque and gravity at -9.81. Maybe I'm having some king of mental block on the topic.

[image]

[image]
[image]

This is the object with the above dims in Solidworks.
This shows the mass, center of mass, and inertial matricies with Kilograms and Meters.

[image]

Thank you.

MotorCityCobra, 2022-02-25:
@yuvaltassa 
Hang on... I think I found another option to save the stl with different units. I'm new to SolidWorks too.
Sorry for being a rookie.
I don't know if this will work...

MotorCityCobra, 2022-02-25:
@yuvaltassa 
Sorry to have tagged you with those comments. I think it might help someone using SolidWorks and dealing with importing meshes in the future.

I have a question about actuators / gears that I don't see in the docs...
What is the formula for the force applied by default with the gear at `1`? Which is a gain of `1`?  
Maybe I missed it but this could be added to the docs. I looked in the C++ and didn't see exactly what the torque is in the code either. Thank you.