google-deepmind/mujoco #159: I'm trying to get a hinge joint to open and close with `mjcb_act_dyn` but the force on the actuator builds up

MotorCityCobra, 2022-02-27:
I'm using `mjcb_act_dyn` and `mjcb_act_gain` to get a model's joint to open and close with a timed function in the C++ file.  
I added the functions to the simulate file and the hinge joint does open and close, but for some reason it is adding the force from every previous movement. Maybe I can explain this way... Instead of dyn = 1 it is doing something like dyn += 1 with every movement. Eventually the model crashes the simulation.  

In the terminal I have `d->actuator_force[0]` printing the value. As seen in the photo, the force is accumulating.  

[image]

[XML and meshes can be downloaded here](https://drive.google.com/drive/folders/1YYG_xZe3_9dEENBbRZFPxDKO1LhEAe-O?usp=sharing)

```
<mujoco model="egg_slicer">
    <compiler angle="radian" meshdir="meshes/"/>
    <size njmax ="1000" nuserdata="600000"/>
    <default class="viz">
        <geom contype="1" conaffinity="1" group="1" type="mesh" rgba=".51 .6 .7 1"/>
    </default>

    <asset>
        <mesh name="base"         file="C:\Users\ooo\Documents\meshes-m2\egg_mash1.stl" />
        <mesh name="swing"         file="C:\Users\ooo\Documents\meshes-m2\egg_mash2.stl" />
    </asset>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/> -->
        <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
        <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    </asset>
    	<option timestep="0.001" gravity="0 0 -9.81"/>

    <worldbody >

        <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>
        <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

        <!-- start building our model -->
        <body >
            <joint type="free" limited='false' damping="0" stiffness="0" armature="0"/>

            <geom name="slicer_base" mesh="base" class="viz"/>
                <inertial pos="0.132891 0.127792 0.104579 " mass="0.060797"
                diaginertia="0.000086 0.000309 0.000248"/>
            <body pos="0 0 0">

                <geom name="slicer_swing" mesh="swing" class="viz"/>
                <joint name="hinge" pos="0.22860842 0.12056147 0.00339329"
                axis="0 0 1"
                limited="true" range="-3.0 0" />
                <inertial pos="0.116117 0.134661 0.104578" mass="0.087188 "
                diaginertia="0.000172 0.000642 0.000491"/>
            </body>
        </body>
    </worldbody>

    <actuator>
        <!-- <motor name="hinge" gear="0.156766" joint="hinge"/> -->
        <general dyntype="user" gaintype="user" biastype="user" gear="0.156766" joint="hinge" name="hinge"/>
    </actuator>
</mujoco>
```

This is the relevant portion of the `simulate` C++ file I modified, which is from the `sample` folder.

```
int gain;
int dyn;

mjtNum gain_callback(const mjModel *m, const mjData *d, int id_01)
{
    return gain;
}
mjtNum dyn_callback(const mjModel *m, const mjData *d, int id_01)
{
    return dyn;
}

// run event loop
int main(int argc, const char** argv)
{
    // initialize everything
    init();

    // request loadmodel if file given (otherwise drag-and-drop)
    if( argc>1 )
    {
        mju_strncpy(filename, argv[1], 1000);
        settings.loadrequest = 2;
    }

    // start simulation thread
    std::thread simthread(simulate);

    // event loop
    while( !glfwWindowShouldClose(window) && !settings.exitrequest )
    {
        // start exclusive access (block simulation thread)
        mtx.lock();

        // load model (not on first pass, to show "loading" label)
        if( settings.loadrequest==1 )
            loadmodel();
        else if( settings.loadrequest>1 )
            settings.loadrequest = 1;

        // handle events (calls all callbacks)
        glfwPollEvents();

        // prepare to render
        prepare();

        // end exclusive access (allow simulation thread to run)
        mtx.unlock();

        // render while simulation is running
        render(window);
        if (m)
        {
            mjcb_act_gain = gain_callback;
            mjcb_act_dyn = dyn_callback; 

            std::cout << d->actuator_force[0] << " ";

            if ((int)((int)d->time % (int)4) == (int)0 || (int)((int)d->time % (int)4) == (int)1)
            {
                dyn = 0;
                gain = -1;
            } else
            {
                dyn = -3;
                gain = 1;
            }
        }
    }

```

yuvaltassa, 2022-02-27:
That's exactly what `mjcb_act_dyn` is. You are adding [activation dynamics](https://mujoco.readthedocs.io/en/latest/computation.html#geactuation) to your model, making your actuator stateful.

closed 2022-02-27 by yuvaltassa

MotorCityCobra, 2022-02-27:
But I don't understand why the actuator force builds up.  

[image]

If I make the bias -1 and 1 and leave the gain as null it seems to be stable. I don't understand it.  
If I don't leave the gain as null with the code below the actuator force builds up again.  

```

int gain;
int dyn;
int bias;

mjtNum gain_callback(const mjModel *m, const mjData *d, int id_01)
{
    return gain;
}
mjtNum dyn_callback(const mjModel *m, const mjData *d, int id_01)
{
    return dyn;
}
mjtNum bias_callback(const mjModel *m, const mjData *d, int id_01)
{
    return bias;
}

// void myController(const mjModel* mArg, mjData* dArg) {
// //Some useful code here
// dArg->act[0] = -100.9;
// }

// run event loop
int main(int argc, const char** argv)
{
    // initialize everything
    init();

    // request loadmodel if file given (otherwise drag-and-drop)
    if( argc>1 )
    {
        mju_strncpy(filename, argv[1], 1000);
        settings.loadrequest = 2;
    }

    // start simulation thread
    std::thread simthread(simulate);

    // event loop
    while( !glfwWindowShouldClose(window) && !settings.exitrequest )
    {
        // start exclusive access (block simulation thread)
        mtx.lock();

        // load model (not on first pass, to show "loading" label)
        if( settings.loadrequest==1 )
            loadmodel();
        else if( settings.loadrequest>1 )
            settings.loadrequest = 1;

        // handle events (calls all callbacks)
        glfwPollEvents();

        // prepare to render
        prepare();

        // end exclusive access (allow simulation thread to run)
        mtx.unlock();

        // render while simulation is running
        render(window);
        if (m)
        {
            mjcb_act_gain = gain_callback;
            mjcb_act_dyn = dyn_callback; 
            mjcb_act_bias = bias_callback;

            std::cout << d->actuator_force[0] << " ";

            if ((int)((int)d->time % (int)4) == (int)0 || (int)((int)d->time % (int)4) == (int)1)
            {
                dyn = 0;
                bias = -1;
            } else
            {
                dyn = -3;
                bias = 1;
            }
        }
    }
```

yuvaltassa, 2022-02-28:
It builds up because the actuator has a _state_. 
That is what `<general dyntype=...` does, it gives your actuator an _internal state_.
This state is "remembering" the previous actions because it is an [integral](https://en.wikipedia.org/wiki/Integral) defined by the first equation in the section "Activation dynamics" [here](https://mujoco.readthedocs.io/en/latest/computation.html#geactuation).

MotorCityCobra, 2022-02-28:
@yuvaltassa 
Thank you.  
Integrals and internal states, huh?  
I'm going to try to learn about this myself and not ask you for the answer outright.  
Could I have a clue? Is it something to do with PID Control?  

This actuator is an unexpected rabbit hole of complexity. My real life stepper motor just has variables of torque, speed, and distance travelled.

MotorCityCobra, 2022-03-02:
@yuvaltassa 
I'm trying to figure it out myself and not ask for a spoon feeding.
I have a question...  

You said "It [the gain] builds up because the actuator has a state."  

I'm dealing with only one actuator and I can't reset the gain to zero.  
I'm calling on this same actuator under three different conditions in a loop. Under two of the three conditions the gain is set to equal `0`. In the last condition the gain is set to `2`.  
Every time it comes back to the condition where the gain was set to `2` the actuator force goes higher. Can I ask why the actuator gain is not reset under this condition?  
The conditions are just intervals within eight seconds.
I know I was pointed to the math and will likely be pointed to the math again, but I don't see why the gain can't be universally set to zero.  
My only goal is to set the gain to something that is not zero and then to reset it, and the state, the force back to zero.
Thank you.

```
        if (m)
        {

            mjcb_act_dyn = dyn_callback; 
            mjcb_act_bias = bias_callback;

            std::cout << d->actuator_force[0] << " ";

            if ((int)((int)d->time % (int)8) == (int)0 || (int)((int)d->time % (int)8) == (int)1
                || (int)((int)d->time % (int)8) == (int)2 || (int)((int)d->time % (int)8) == (int)3)
            {
                dyn = 0;
                bias = -10;
                gain = 0;
            } else
            {

                if ((int)((int)d->time % (int)8) == (int)6 || (int)((int)d->time % (int)8) == (int)7)
                {
                    dyn = -3;
                    bias = 10;
                    gain = 2;
                    std::cout << "With gain" << "\n";

                } else
                {

                dyn = -3;
                bias = 10;
                gain = 0;
                }
            }
        }
```