google-deepmind/mujoco #160: how to visualize the joints and frames in mujoco-py?

yinzikang, 2022-03-02:
there are ways to visualize the joints and frames in mjvOption in mujoco of cpp. How can I find the corresponding ways in the mujoco-py?

kevinzakka, 2022-03-04:
Are you using dm_control or mujoco-py? mujoco-py is unrelated and developed by OpenAI.

If you're using [dm_control](https://github.com/deepmind/dm_control)'s bindings, you can do something like:

```python
# The basic mujoco wrapper.
from dm_control import mujoco

# Access to enums and MuJoCo library functions.
from dm_control.mujoco.wrapper.mjbindings import enums
from dm_control.mujoco.wrapper.mjbindings import mjlib

scene_option = mujoco.wrapper.core.MjvOption()
scene_option.frame = enums.mjtFrame.mjFRAME_GEOM
scene_option.flags[enums.mjtVisFlag.mjVIS_JOINT] = True
pixels = physics.render(scene_option=scene_option)
```

There's a really nice intro tutorial in their [Colab notebook](https://colab.research.google.com/github/deepmind/dm_control/blob/master/tutorial.ipynb) and you can always look things up in the API reference of their [documentation](https://mujoco.readthedocs.io/en/latest/APIreference.html).

yinzikang, 2022-03-04:
oh I see, this really helps. Thanks

closed 2022-03-04 by yinzikang

richardrl, 2024-12-27:
Is there any way to selectively turn frame visualizations on and off in the GUI? Like in Rviz

yuvaltassa, 2024-12-28:
> Is there any way to selectively turn frame visualizations on and off in the GUI? Like in Rviz

Yes. Watch [this](https://mujoco.readthedocs.io/en/latest/programming/samples.html#sasimulate)