google-deepmind/mujoco #161: How to send cartesian target for the end effector ?

aled96, 2022-03-02:
Hi, I am new to MuJoCo and I would like to use it to push an object with a robotic arm, checking how the object moved in the end.

I loaded successfully the model and I have the callbacks for the mouse to move the camera.

Now I have some questions:
1- How can I control the joints in order to remain still ? Now the arm is oscillating due to gravity I think.
2- How can I send a cartesian target to reach with the end-effector ?
3- How can I get the position of an object in the space to check the resulting pose of the object after a push ?

Thank you for the help !

yuvaltassa, 2022-03-02:
1. Add damping to your joints.
2. This is not implemented as a native transmission type, though we are considering doing it. To do it yourself, use the Jacobian of the thing you are trying to control (accessed with `mj_jac()`) to transform from Cartesian space to joint space.
3. The global positions of everything are in `mjData` with the prefix 'x', e.g. the global positions of geoms are in `geom_xpos`

closed 2022-03-02 by yuvaltassa

aled96, 2022-03-03:
I added the damping and I am trying yo control the arm to reach a position.
I have defined the "actuators" for the joints I need to control.

I have the desired joint angles (in m->qpos0), now I do not understand how to send them.
I tried to set d->ctrl = des_j_ang but it doesn't reach the position I want. In addition, I tried also to use the PD controller in the following way:

```
for(int i = 0; i < m->nu; i++){
        if (d->time - last_update > 1.0/60.0)
        {
            mjtNum vel = (d->qpos[i] - m->qpos0[i])/(d->time-previous_time);
            ctrl = 0.5*(vel-5.0*d->qpos[i]);
        }
        
        d->ctrl[i] = ctrl;
    }
    
    if (d->time - last_update > 1.0/60.0)
    {
        last_update = d->time;
        previous_time = d->time;
    }
```

aled96, 2022-03-03:
Thanks, I've already set the actuators as position with the kp gain, but the arm is not moving even changing d->ctrl in the controller written above.

yuvaltassa, 2022-03-03:
This should work.
Increase `kp`?

aled96, 2022-03-03:
I set `kp=50` but nothing changes. I noticed that, even if the robot starts in the desired position, `d->qpos` is different from `m->qpos0`. Since I am using `d->qpos` to set `d->ctrl`, may it be related to this ?

yuvaltassa, 2022-03-03:
Can you please load your model in `simulate` and see what happens when you tweak the actuator sliders?
If that doesn't clarify, please post your model here and I'll take a look.

aled96, 2022-03-03:
> Can you please load your model in `simulate` and see what happens when you tweak the actuator sliders?

I tried to move the slider on the "Control" panel but the robot is not moving.

However, I noticed that if in the code I incrementally increase `d->ctrl` the robot moves (of course, not as I want).

aled96, 2022-03-03:
The model is the following:  [centauroTEST.zip](https://github.com/deepmind/mujoco/files/8177695/centauroTEST.zip)
 
 I obtained using an urdf model, then I added the actuators definition.

aled96, 2022-03-03:
I changed the code from:
```
for(int i = 0; i < m->nu; i++){
        if (d->time - last_update > 1.0/60.0)
        {
            mjtNum vel = (d->qpos[i] - targeti])/(d->time-previous_time);
            ctrl = 0.5*(vel-5.0*d->qpos[i]);
        }
        d->ctrl[i] = ctrl;
    }
    
    if (d->time - last_update > 1.0/60.0)
    {
        last_update = d->time;
        previous_time = d->time;
    }
```

to this:
```
for(int i = 0; i < m->nu; i++){
        if (d->time - last_update > 1.0/60.0)
        {
            mjtNum vel = (d->qpos[i] - targeti])/(d->time-previous_time);
            ctrl = 0.5*(vel-5.0*d->qpos[i]);
            last_ctrl_sent = ctrl;
        }
        d->ctrl[i] = last_ctrl_sent;
    }
    
    if (d->time - last_update > 1.0/60.0)
    {
        last_update = d->time;
        previous_time = d->time;
    }
```

And in this case I am able to move only the first joint, the others remain still in the starting position.

yuvaltassa, 2022-03-03:
Looking at your model (which you posted and then deleted?), your joints have huge stiffness. The stiffness is in the same units as `kp`, these are both springs. So you have a spring with strength 5 fighting with a spring with strength 900...

aled96, 2022-03-03:
Sorry, I deleted by mistake !
I increased `kp` and it works but just with one joint, if I change more values it does not move correctly

aled96, 2022-03-04:
I don't get why d->qpos is the following:

```
0.0103655  0.00579894  0.0194636  -0.00926653  0.0191179  -0.00126138  0.00330963  
8.70175e-06  0.00576991  -0.0133269  0.0091694  0.019168  0.00123955  0.00331838  -7.39627e-06
```

When the qpos0 (and the actual position of the robot) is:
```
0     -0.520149     -0.320865       -0.274669      2.23604      -0.0500815       0.781461      0.0567608   
   -0.520149     0.320865      0.274669       2.23604      0.0500815      0.781461      -0.0567608
```

How can I check the actual joint angles so that I can correctly set `d->ctrl` ?

yuvaltassa, 2022-03-05:
d->qpos are the joint angles.
It's not clear what your issue is and I don't have a model to look at, so not sure how to help.

aled96, 2022-03-05:
Yes but looking at d->qpos, I see the following values even if the robot configuration I see is different...

> 0.0103655  0.00579894  0.0194636  -0.00926653  0.0191179  -0.00126138  0.00330963  
> 8.70175e-06  0.00576991  -0.0133269  0.0091694  0.019168  0.00123955  0.00331838  -7.39627e-06

yuvaltassa, 2022-03-05:
You can save these numbers in a [keyframe](https://mujoco.readthedocs.io/en/latest/XMLreference.html#keyframe)  in your XML and load the keyframe in `simulate` as a debugging step to try to understand what's happening.

Remember that the position you are seeing in simulate corresponds to the geom_xpos of the previous step, not the current step, since the kinematics of the current step have yet to be computed, but this usually only corresponds to a very small discrepancy.